koalazak/dorita980 #1: Finding username and password suggestion

sheppy99, 2016-06-28:
Hi,
I've just came across your work on the Roomba 980 and have an idea about how to get an additional username / password added. Last night I added a second iphone to my Roomba and this may be the way in for dorita980.
On the new phone, the app finds the Roomba on the normal wifi network and there is a button in the app that says something like "add Roomba" clicking this gives a prompt to press one of the buttons on the Roomba, once this is done the app works presumably with a new username and password. Could this data interchange be emulated so the software asks for a new "app user" from the Roomba device? It would be easier than sniffing the traffic?
Cheers
Kevin

koalazak, 2016-06-28:
Hi,
sounds good! I have no idea about this method to add new phone. I will research on this way! but the first problem I see is the attempt to sniff trafic between the phone and the roomba, because it seems Roomba is doing a [SSL Key Pinning](https://en.wikipedia.org/wiki/HTTP_Public_Key_Pinning) to protect agains MITM attacks. So I can not sniff the trafic between the phone and the roomba.

The only way I see is decompile the .apk from the android app and see wich urls and commands the app send to the roomba https server. I try that but only found the commands and urls to the cloud, not to the roomba. We should look the code in deep to found that and then emulate that communication ;)
thanks!

koalazak, 2016-09-16:
Thanks @sheppy99! Now the npm script `npm run getpassword` is ready and working to get Roomba username and password. Take a look to readme file :)

closed 2016-09-16 by koalazak

sheppy99, 2016-09-16:
Fantastic work, will give it a try over the weekend

sheppy99, 2016-09-30:
@koalazak I've finally got around to following your instructions and I have a username and password for my Roomba. I've also copied the start cleaning into a file I've named start.js in the dorita980 folder.
I then tried to run it with "node start.js" and I get the error below - I'm a newbie to node.js so I think I'm doing something obvious wrong?

Error: Cannot find module 'dorita980'
    at Function.Module._resolveFilename (module.js:455:15)
    at Function.Module._load (module.js:403:25)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/data/roomba/dorita980/start.js:1:79)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)

koalazak, 2016-09-30:
@sheppy99 If you want use dorita980 in your own project, you need to install dorita980 using `npm` command on your project directory (where you have index.js):

```
cd your_project
npm install dorita980 --save
node index.js
```

`npm install` will create a `node_modules` forder with all modules used by your project. If you want to use other npm modules just install them with `npm install <name>`.
The `--save` option just create and store this info in a package.json file for your project.
You can read more here: https://docs.npmjs.com/files/package.json

`require()` just search for modules in node_modules folder or in current path. So if dorita980 is not installed, throw Error: Cannot find module dorita980.

regards!

sheppy99, 2016-10-01:
Thanks, after a little experimentation I managed to get it to work on my Windows PC, now to move it to the Pi & Openhab.
A minor error I think is in your example "Quick start via local request on your LAN" I think the second line should read

     var myRobotViaLocal = new dorita980.Local('MyUsernameBlid', 'MyPassword', '192.168.1.104'); // robot IP address 

as 

     var myRobotViaCloud = new dorita980.Local('MyUsernameBlid', 'MyPassword', '192.168.1.104'); // robot IP address

throws an error.

koalazak, 2016-10-02:
thanks! i fix that!