koalazak/dorita980 #5: Thank you + Questions

mtwhitley, 2016-11-03:
Thank you so much for creating this library. I do have a few questions after getting it running:

1. Does the getMission local function need to be called once - after the mission or more than once while the robot is cleaning? 
2. Does the getMission local function return an array of values or a single value?
3. The library does not appear to include the code to generate the cleaning map. Is this something you would be willing to also share?

Thank you again for contributing the library. I very much appreciate it!

koalazak, 2016-11-03:
Hi @mtwhitley,

1) You can call the getMission function at any time. This method shows the current mission status. If you call then when robot is charging you see diferent 'phase' value than if you calle the method when robot is cleaning. This method shows de X and X position in the moment you call. So call more than once while the robot is cleaning to get the X,Y position every 800ms for example to draw a map.
2) getMission() returns a JSON object with multiple properties. here an example while charging:
```
{ ok:
   { flags: 0,
     cycle: 'none',
     phase: 'charge',
     pos: { theta: 179, point: {x: 0, y: 0} },
     batPct: 99,
     expireM: 0,
     rechrgM: 0,
     error: 0,
     notReady: 0,
     mssnM: 0,
     sqft: 0 },
  id: 2 }
```
and here an example while cleaning (note `cycle` and `phase` properties)
```
{ ok:
   { flags: 4,
     cycle: 'quick',
     phase: 'clean',
     pos: { theta: 179, point: {x: 150, y: 22} },
     batPct: 70,
     expireM: 0,
     rechrgM: 0,
     error: 0,
     notReady: 0,
     mssnM: 0,
     sqft: 0 },
  id: 4 }
```
3) Im working on a more universal way to draw the map (rigth now its very hacerdcoded to work with my home dimensions). I think it will be ready soon and I will publish it on [rest980](https://github.com/koalazak/rest980) repo as experimental stuff, so follow up rest980! :)

thanks you!

closed 2016-11-03 by koalazak

mtwhitley, 2016-11-03:
Thank you @koalazak for the explanation. This makes more sense now. 

I have followed the rest980 repo per your guidance. I might also try my hand at a plotting library. I'm looking forward to giving your work a try.

I do have one more question for you. Do you know if the HTTP APIs support reading sensor data or sending any of the OpCodes from the Open Interface spec?

Thanks again,
Michael

koalazak, 2016-11-03:
afaik the api on the robot or in the cloud doest not support sensor reading or sending opcodes. Maybe connecting via serial port directly to the robot it is possible, but i dont know.

best regards,