koalazak/dorita980 #6: Flags for getMission()

koalazak, 2016-11-28:
Following https://github.com/koalazak/rest980/issues/3
Open this thread to organize Reverse engineering over `flags` field in `getMission` method.

jtscott, 2016-12-11:
It took me a while to get the specific bin full situation, but here is what I have right now:
`robot_status: '{"flags":1,"cycle":"none","phase":"charge","pos":{"theta":162,"point":{"x":128,"y":242}},"batPct":100,"expireM":0,"rechrgM":0,"error":0,"notReady":16,"mssnM":197,"sqft":534}',`

[And here is a screenshot from the roomba app of the cleaning ](https://dl.dropboxusercontent.com/s/8uv7q7oay4h1cmw/Photo%202016-12-11%2C%2010%2020%2001%20AM.png?dl=0)

jernejs, 2016-12-20:
I have iRobot Home app showing "Bin: Overdue", and had flags 4; after cleaning the bin, flags showed 0, but the app was still showing Overdue, and stayed like that until I selected Reset Status in the app.

jernejs, 2016-12-21:
I looked into this a bit more, and it appears that flags & 1 means that bin is full and flags & 2 means bin removed. I'm not certain what 4 means, but it could mean battery percentage under 50%.

jernejs, 2016-12-21:
More info: still not sure what 4 means (seems to get set at random times), but 8 is apparently set when Roomba is playing a sound (either an error message, or the start/stop chime).

koalazak, 2016-12-21:
Its a little tricky cos only one decimal number determine multiples values for multiples things at the same time.
I thing 4 is when Roomba is recently wakeup after iddle time.
Here my research:
(Can you confirm this table with your robot?)

| iddle | Bin removed | Beeping | Bin Full |  | Flag | Bits | 
|-------|-------------|---------|----------|--|------|------| 
| no    | no          | no      | no       |  | 4    | 0100 | 
| no    | no          | no      | yes      |  | 5    | 0101 | 
| no    | no          | yes     | no       |  | 12   | 1100 | 
| no    | no          | yes     | yes      |  | 13   | 1101 | 
| no    | yes         | no      | no       |  | 6    | 0110 | 
| yes   | no          | no      | no       |  | 0    | 0000 | 
| yes   | no          | no      | yes      |  | 1    | 0001 | 
| yes   | yes         | no      | no       |  | 2    | 0010 | 
| yes   | no          | yes     | no       |  | 8    | 1000 | 

Now i will try to correlate each state with one bit. And then calculate the Bitmask to read each state individually.

I dont know where is the flags for other states like rollers are blocked, detecting dirty spots, when clean chair legs or border the house, etc.
Can someone help you find these? :)

thanks!

jtscott, 2016-12-21:
Hi Guys, my real world scenarios have been held up because of the holidays and people being home not running the roomba. I think I am close to having sensors/rollers blocked... hope to have that soon!

jernejs, 2016-12-22:
I'm writing my own utility in perl, here's what I learned so far:

flags:
  Bit 1 = bin full
  Bit 2 = bin removed
  Bit 3 = ??
  Bit 4 = playing sound
Haven't seen any other bits set; bit 3 seems to get set around the time Roomba docks - could mean docking, but it doesn't get reset until some time after it's docked.

cycle and phase tell you what Roomba is doing (cycle: none, quick, spot, dock; phase: charge, run, stuck, hmPostMsn [docking after cleaning finished], hmUsrDock [pressed the home button or sent dock command]). I haven't noticed any special status when it's doing edge cleaning.

error is set when Roomba is in stuck phase; error numbers seem to match the numbers from manual, but for the wheels the code in manual seems to match right wheel, while left wheel has it's own code (I got error 4 when Roomba got stuck under a couch and couldn't move left wheel).

notReady is set when Roomba is stopped, and can't start:
	0 => 'Ready',
	2 => 'Both wheels dropped',
	3 => 'Left wheel dropped',
	4 => 'Right wheel dropped',
	7 => 'Bin missing',
There are probably others, these are easy to reproduce.

jernejs, 2016-12-22:
One more value for notReady: 1 - near a cliff (I've seen this get set while running, then cleared when Roomba turned around)

jtscott, 2016-12-22:
I had flag 12 today when the roomba got stuck, no other statuses or beeping...?

robot_status: '{"flags":12,"cycle":"none","phase":"stuck","pos":{"theta":-88,"point":{"x":149,"y":-301}},"batPct":3,"expireM":0,"rechrgM":0,"error":18,"notReady":15,"mssnM":77,"sqft":289}',

jernejs, 2016-12-22:
According to manual error 18 means Roomba can't return to the dock. Haven't seen notReady 15 yet, but since your batPct is 3, I'm assuming it means low battery. Not sure why you got flags 12 though.

Has anybody figured out what expireM and rechrgM are? Never seen anything other than 0 with them; also, what exactly does sqft show - I see it increase and decrease while vacuuming, so it can't be purely area.

koalazak, 2016-12-22:
@jtscott Not sure why you got flags 12 though. This value remains a while? or just a second?.

Hey @jernejs! thanks for your help! 
`rechrgM` is the time to left to recharge the batery. 
`expireM` is the time to left to restart a clean job after first clean lap (when roomba go home to charge and continue later).
`sqft` is the area in Square foots cleaned in this mission.

jernejs, 2016-12-22:
OK, so I assume rechrhM and expireM are only set if Roomba didn't finish cleaning, but had to go recharge first?

I see sqft jump around quite a lot while cleaning (going up and down) - any ideas why?

I've seen flags 12 myself while Roomba was playing a sound, but it always returned to 4 after it was finished playing.

jtscott, 2016-12-22:
@koalazak the roomba got stuck (and still is stuck), flag 12 is persisting. Here is the full roomba_status dump:

`
{ status: 'OK',
  method: 'getStatus',
  sku: 'R98----',
  country: 'CA',
  postalCode: 'T3C2Z1',
  regDate: '2016-09-24',
  hardwareVersion: 0,
  manualUpdate: 0,
  swUpdateAvailable: '',
  schedHold: 0,
  autoEvacCount: '',
  autoEvacFlags: '',
  wifiDiagnostics: '',
  autoEvacModel: '',
  milestones: '',
  robotName: 'Roomba',
  robotLanguage: 0,
  tzName: 'America/Edmonton',
  twoPass: 1,
  noAutoPasses: 1,
  openOnly: 0,
  carpetBoost: 0,
  binPause: 0,
  vacHigh: 0,
  noPP: 0,
  ecoCharge: 0,
  mission: '{"nMssn":74,"done":"cncl","flags":1025,"sqft":289,"runM":69,"chrgM":0,"pauseM":8,"doneM":0,"dirt":0,"chrgs":0,"saves":1,"evacs":0,"pauseId":0,"wlBars":[100,0,0,0,0]}',
  preventativeMaintenance: '[{"partId":"bin","date":"2016-11-26","distance":30160,"runtime":3026,"months":2,"notified":false},{"partId":"core","date":"2016-11-26","distance":30160,"runtime":3026,"months":2,"notified":false},{"partId":"extractor","date":"2016-11-26","distance":30160,"runtime":3026,"months":2,"notified":false}]',
  cleanSchedule: '{"cycle":["none","none","none","none","none","none","none"],"h":[9,9,9,9,9,9,9],"m":[0,0,0,0,0,0,0]}',
  robot_status: '{"flags":12,"cycle":"none","phase":"stuck","pos":{"theta":-88,"point":{"x":149,"y":-301}},"batPct":3,"expireM":0,"rechrgM":0,"error":18,"notReady":15,"mssnM":77,"sqft":289}',
  softwareVersion: 'v1.6.6',
  lastSwUpdate: '2016-09-25 09:28:59+0000',
  engBuild: 0,
  bbrun: '{"hr":73,"min":48,"sqft":210,"nStuck":47,"nScrubs":18,"nPicks":256,"nPanics":419,"nCliffsF":939,"nCliffsR":12715,"nMBStll":2,"nWStll":20,"nCBump":1}',
  missing: false,
  ota: '{"st":0,"err":0,"lbl":""}',
  readyToMigrate: 1 }
`

jernejs, 2016-12-22:
@jtscott Are you using the cloud API or local API? I only use the local one.

jtscott, 2016-12-22:
@jernejs yeah I am using the cloud api

jernejs, 2016-12-22:
That may explain why the flags don't change.

koalazak, 2016-12-23:
yes @jtscott, try using local api what flag you see.
yes @jernejs assume that.

koalazak, 2016-12-30:
pushed (v2.5.1) what we have so far: https://github.com/koalazak/dorita980#getmissionautodecodeflags

koalazak, 2017-02-14:
not needed in new firmware 2.0.0

closed 2017-02-14 by koalazak

idoodler, 2019-12-05:
I integrated my Roomba e5 in my Loxone Smart Home, I want to obtain the current state of the Roomba like Stucked, Docked, Working, ...

However I am unable to find the `flags` property in the `getMission()` function. Has this property been removed?

I am using [rest980](https://github.com/koalazak/rest980) on a Raspberr Pi, my Roomba is running `3.4.46-76`