koalazak/dorita980 #9: Use of const in strict mode

mcmurm, 2016-12-29:
First, nice job on this library. Looks very promising. Here's one issue I ran into that might help others exploring your work. I'm not an expert, but I think this is due to versioning difference in node.js?

`hass@sob:~/roomba/dorita980$ node ./bin/getpassword.js "<IP>"

/home/hass/roomba/dorita980/bin/getpassword.js:3
const request = require('request');
^^^^^
SyntaxError: Use of const in strict mode.
    at Module._compile (module.js:439:25)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:902:3
`

Fixed by using the --harmony node.js switch:

`hass@sob:~/roomba/dorita980$ node --harmony ./bin/getpassword.js "<IP>"
Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light. Then wait and look here...
========>
Good job!
Password: *****************
Username/blid: *******************
Use this credentials in dorita980 lib :)
`

koalazak, 2016-12-29:
hi, what node version you have?

mcmurm, 2016-12-29:
hass@sob:~/roomba$ node --version
v0.10.25

koalazak, 2016-12-29:
dorita980 is not compatible with node < 4.0. Tests [are passing](https://github.com/koalazak/dorita980/blob/master/.travis.yml) ok for node 4, 5, 6 and latest.
Basicaly, dorita980 use `Promises` and they are [introducing](http://node.green/) on node 0.12

`const` are supported in node 0.10 but with harmony flag.

I will update the readme to 0.10 users to advise about harmony flag only for getpassword.js file (the rest of dorita980 will not work anyaway cos promises)

thanks!

koalazak, 2016-12-30:
pushed in https://github.com/koalazak/dorita980/commit/7f52888e6a9aa9440fed41e322e66f099236da7b

closed 2016-12-30 by koalazak

mcmurm, 2016-12-30:
Awesome, thanks, and good work.

mcmurm, 2016-12-30:
FYI, I updated to node 6.x and everything is working fine. This also fixed rest980 (wouldn't run under node .10...) Thanks again.