koalazak/dorita980 #10: firmware 2.0.0-34 ready?

koalazak, 2016-12-30:
What about this new firmware update? 
Does anyone here upgrade their roomba?
I heard that the new firmware no longer listens on port 443.
For now I recommend deny internet access to your roomba to keep v1.6.6 and enjoy dorita980.

TheMasterBaron, 2017-01-12:
Mine stopped working, and yup, It's firmware 2.0.0-34.  :(

xibriz, 2017-01-12:
Does the roomba upgrade automaticly?

koalazak, 2017-01-12:
yes @xibriz. but 2.0.0-34 seems to be a beta version only for few users. on 2.0.0. will be update for everyone.

koalazak, 2017-01-12:
Hi @TheMasterBaron thanks for reply!

please, can you run this code and share the output?

test.js
```javascript
const dgram = require('dgram');

const server = dgram.createSocket('udp4');

server.on('error', (err) => {
  server.close();
  console.log(err);
});

server.on('message', (msg) => {
  try {
    var parsedMsg = JSON.parse(msg);
    console.log(parsedMsg);
    process.exit();
  } catch (e) {}
});

server.on('listening', () => {
  console.log('Looking for robots...');
});

server.bind(5678, function () {
  const message = new Buffer('irobotmcs');
  server.setBroadcast(true);
  server.send(message, 0, message.length, 5678, '255.255.255.255');
});
```
Is something like this:
```
$ node test.js
Looking for robots...
{ ver: '2',
  hostname: 'Roomba-xxxxxxxxxxxxxx',
  robotname: 'Dorita',
  ip: '192.168.1.104',
  mac: 'xx:xx:xx:xx:xx:xx',
  sw: 'v1.6.6',
  sku: 'R98----',
  nc: 0,
  proto: 'http' }
```

thanks!

KleemannT, 2017-01-12:
Output from test.js:

{ ver: '2',
  hostname: 'Roomba-xxxxxxxxxxxx',
  robotname: 'Heike',
  ip: '192.168.1.15',
  mac: 'xx:xx:xx:xx:xx:xx',
  sw: 'v2.0.0-34',
  sku: 'R98----',
  nc: 0,
  proto: 'mqtt' }

koalazak, 2017-01-12:
```proto: 'mqtt'```
oh crap...

koalazak, 2017-01-12:
@KleemannT thanks!
can you test one more thing please?

cou can try deny the internet connection to your roomba from the router (parent control or something), restar the roomba and try to send commands to look if is listen on port 443 or 80?
(maybe if new firmware detect no internet connection to the cloud, then open local API, i dont know.)

thanks!

KleemannT, 2017-01-13:
internet access locked for robot
no connection with dorita980
portscan against robot from 1-65535
open port on TCP:8883

koalazak, 2017-01-13:
thank you @KleemannT !
sounds like the new firmware is no longer using http/s portocol and now use `mqtt` protocol on port 8883. MQTT is more apropiated for IoT things, btw.
When my roomba updates her firmware I'm going to have a lot of work to do in dorita980 ¬¬

thanks!

CLARENNE-Q, 2017-01-25:
Hi everybody,

First of all, thanks for this good documentation/code it's really appreciated !

I have the same issue:

`node index.js 
Looking for robots...
{ ver: '2',
  hostname: 'Roomba-30F88508216136XX',
  robotname: 'Terminator',
  ip: '192.168.2.22',
  mac: '80:A5:89:AA:DB:XXX',
  sw: 'v2.0.0-34',
  sku: 'R98----',
  nc: 0,
  proto: 'mqtt' }
`

> sudo nmap -sU -p 137,5353 --script nbstat,dns-service-discovery 192.168.2.22

Starting Nmap 7.40 ( https://nmap.org ) at 2017-01-24 22:59 EST
Nmap scan report for 192.168.2.22
Host is up (0.047s latency).
PORT     STATE  SERVICE
137/udp  closed netbios-ns
5353/udp closed zeroconf
MAC Address: 80:A5:89:AA:DB:04 (AzureWave Technology)

Nmap done: 1 IP address (1 host up) scanned in 0.55 seconds

Is it possible to make a downgrade ?

koalazak, 2017-01-25:
Hi @CLARENNE-Q, sounds like you have a firewell or something (and port is 8883 not default 1883). Try knoking with netcat command:
```
nc -vz 192.168.2.22 8883
```

I dont know if its possible to make a downgrade. If you found a way for that, please come back and share how to :)

regards,

IngmarStein, 2017-02-01:
@koalazak, has your Roomba updated to 2.0.0-34 already?

Here's the netcat output for an updated robot:
```
$ nc -vz <ip> 8883
found 0 associations
found 1 connections:
     1:	flags=82<CONNECTED,PREFERRED>
	outif en0
	src 192.168.178.20 port 51423
	dst 192.168.178.22 port 8883
	rank info not available
	TCP aux info available

Connection to <ip> port 8883 [tcp/*] succeeded!
```

CLARENNE-Q, 2017-02-02:
Hi all,

Sorry, I'am late:

`$ nc -vz 192.168.2.22 8883
found 0 associations
found 1 connections:
     1:	flags=82<CONNECTED,PREFERRED>
	outif en0
	src 192.168.2.14 port 51580
	dst 192.168.2.22 port 8883
	rank info not available
	TCP aux info available

Connection to 192.168.2.22 port 8883 [tcp/*] succeeded!`

koalazak, 2017-02-02:
Hi guys,

@IngmarStein no yet ;/
@CLARENNE-Q you firewall detects tehe nmap, but you can connect with nc to port 8883.

im in conditions to confirm that the new firmware is no longer using http/s protocol and now use mqtt protocol on port 8883. Dorita980 needs changes to work with 2.0 formware.

Im waiting the update in my robot...

sputt, 2017-02-02:
I have the 2.0.0-34 firmware as of 1/29/2017.  Let me know if there's anything you want me to try

iosdeveloper, 2017-02-03:
Does this change only affect local or also cloud API?

LordLiverpool2, 2017-02-03:
I am the latest victim of this unwanted update. 

If you need me to do any tests, let me know.

normanth, 2017-02-04:
Hi,

I have the same problem.

With "local" access I get "Internal server error".

With "cloud" access I get infos - but Roomba doesn't act.

http://ds:3000/api/cloud/info/status
{"status":"OK","method":"getStatus","sku":"R980040","country":"",........

http://ds:3000/api/cloud/action/start
{"status":"OK","method":"multipleFieldSet"}

Thanks for the api - very good solution!

Norman

koalazak, 2017-02-04:
thanks guys! 
my roomba is updated now. Im reverse engineering the new api.
Will keep you informed!

sheppy99, 2017-02-04:
Mine has updated and stopped talking to the API. Really with iRobot would stop stuffing about and release a supported API. Not everyone wants several bloated apps on their phone to work stuff when one can do all!!!

Happy to help get this thing working again

JensBonse, 2017-02-06:
I did some sniffing and 2.0.0-34 is for example now connecting to
https://sl7awxqj03.execute-api.us-east-1.amazonaws.com/dev/v1/30F8431482042630/missionhistory?app_id=IOS-12345678-1234-1234-1234-123456789012
And the whole authentication and header stuff looks more tricky. At least not like the previous headers that could be decoded on first sight. 

So seems it is really some beta test. I do not know whether it makes sense to decode everything now.

sheppy99, 2017-02-07:
@JensBonse did you see the app taking to that address and then that address talking to the Robot? I've been using the Wireshark built into my Router to try and see what is happening. I see a connection to the server, but nothing to the robot despite it starting and stopping from the App. I wonder whether on WiFi the connection is going direct from ios app to robot, maybe via broadcast and the app reports the status to the server.
I am new to Wireshark so could of course be wrong!

JensBonse, 2017-02-07:
@sheppy99 I saw the App communicating with that address. I did not sniff the the connection of the cloud service with the robot. I just used a man in the middle attack with a proxy. The proxy prevents the direct connection of the phone to the robot. In general there are two ways to communicate with the robot.
1. Using the cloud as described.
2. Direct connection of the APP to the robot. (I did not sniff this)

sheppy99, 2017-02-07:
This has given me an idea. Tomorrow I will put my iPhone on 5Ghz and the Robot will be on 2.4GHz. I will see if I can capture any app to robot traffic. The router will have to pass it between the 2 radio bands, so maybe I can log it

koalazak, 2017-02-07:
Hi guys, im focusing on the local mqtt api, but talking about cloud i see this flow:
1) Discovery
```
GET /v1/robot/discover/MY_ROBOT_ID HTTP/1.1
Host: disc-prod.iot.irobotapi.com
User-Agent: aspen/1.9.1.184.1 CFNetwork/808.2.16 Darwin/16.3.0
```
response:
```
HTTP/1.1 200 OK
Content-Type: application/json
x-amzn-RequestId: 2c537002-eb08-11e6-aa52-a1421888806e
X-Amzn-Trace-Id: Root=1-58961e0d-00fcdf8067e8fb1f1588e6a5
X-Amz-Cf-Id: f_base64string
{
    "awsRegion": "us-east-1",
    "discoveryTTL": 86400,
    "httpBase": "https://d1nokieziam3wp.cloudfront.net",
    "httpBaseAuth": "https://bcgqn5awqe.execute-api.us-east-1.amazonaws.com/dev",
    "iotTopics": "$aws",
    "irbtTopics": "v005-irbthbu",
    "mqtt": "a2uowfjvhio0fa.iot.us-east-1.amazonaws.com",
    "svcDeplId": "v005"
}
```
this tells where is the cloud api and mqtt cloud servers (there are two mqqt brokers one inside roomba and other in the cloud)

2) login:

```
POST /v1/login HTTP/1.1
Host: d1nokieziam3wp.cloudfront.net get this from discovery
Content-Type: application/json

{"associations":{"0":{"robot_id":"MY_ROBOT_ID","deleted":false,"password":":1:1486234474:ONE_PASSOWRD"}},"app_id":"IOS-UUID"}
```
response:
```
HTTP/1.1 200 OK
Content-Type: application/json
Content-Length: 38
Connection: keep-alive
Date: Sat, 04 Feb 2017 19:07:58 GMT
X-Cache: Miss from cloudfront
X-Amz-Cf-Id: base64string

{"associations": {"0": "ErrorTooNew"}}
```
aparently the password change all the time, maybe is generated using a timestamp (the first numers in password field are a timestampt)

(I dont know what is the correct response for that login. i always get errorTooNew)

Talking about Local API, I use arpspoof to tells to the mobile app that my computer is the roomba, so the trafics pass throw me and then I redirect the trafic with ip forwarding to the roomba.
```
arpspoof -i at0 -t mobile_id roomba_ip
```

thanks for the help!

sheppy99, 2017-02-07:
By putting the phone and Roomba on separate channels I've just managed to sniff the traffic using my Fritzbox. I can see SSL 8883 traffic direct between my iPhone and Roomba. So it seems Roomba and the app talk directly when on the local LAN. Can't see a lot else but thought this may help.

koalazak, 2017-02-07:
Are you using `sslsplit` to see the trafic content? i see some mqtt packages but I dont found the correct configurations to connect with the local broker.
That is what i have for now but i cant connect and listen for topics :/

```
const mqtt = require('mqtt');

const robotID = 'myROBOTID';
const robotIP = 'myROBORIP';
const myROBOTPASS = 'robotpass';

const url = 'mqtts:/' + robotIP; // what protocol? mqtts? tls? tcp?

var options = {
  port: 8883,
  clientId: robotID, // use robot id? or random generated?
  rejectUnauthorized: false,
  protocolId: 'MQTT', // MQTT or MQIsdp?
  protocolVersion: 4, //4 or 3?
  clean: true, // true or false?
  keepalive: 0,
  will: { retain: true,
     qos: 3,
     topic: robotID, //sure?
     payload: ':1:1486234474:' + myROBOTPASS
  }
};

var client = mqtt.connect(url, options);

client.on('error', function (e) {
  console.log('eeeeerrr', e);
});

client.on('close', function () {
  console.log('on close');
});

client.on('disconnect', function () {
  console.log('on disconnect');
});

client.on('reconnect', function () {
  console.log('on reconnect');
});

client.on('offline', function () {
  console.log('on offline');
});

client.on('packetreceive', function (packet) {
  console.log('packet', packet);
});

client.on('connect', function () { // When connected

  console.log('connectado!');
  // the topic is the robot ID? or other?
  client.subscribe(robotID, function () {
    client.on('message', function (topic, message, packet) {
      console.log("Received '" + message + "' on '" + topic + "'");
    });
  });

  // publish a message to a topic
  // client.publish('hello/world', 'my message', function() {
  //  console.log("Message is published");
  //   client.end(); // Close the connection when published
  // });
});
```

akpotter, 2017-02-09:
@koalazak 
I got successful login:

request:(password looks like not change every time)
```
POST /v1/login HTTP/1.1
Host: d1nokieziam3wp.cloudfront.net get this from discovery
Content-Type: application/json

{"associations":{"0":{"robot_id":"MY_ROBOT_ID","deleted":false,"password":":1:1486234474:ONE_PASSOWRD"}},"app_id":"IOS-UUID"}
```

response: (every keys changes every time)
```
{"associations": {"0": "Success"}, "credentials": {"SecretKey": "GIVED_40bits_KEYS", "SessionToken": "GIVED_1248bits_KEYS", "Expiration": "2017-02-09T04:06:23+00:00", "CognitoId": "us-east-1:a8999d99-f295-4532-990d-d8f06d263138", "AccessKeyId": "GIVED_40bits_KEYS"}}{
}
```

iosdeveloper, 2017-02-11:
Would donating a few $ help the effort to update this library? Really bummed by the update and need an API again.

koalazak, 2017-02-12:
nice @akpotter! i can too. Maybe is better put the focus on cloud api now and continue with the local mqtt api later. Im stuck.

@iosdeveloper donations are welcome but honestly they dont burst the process.
botcoin address for donations: 189JxfTBWXMDZomGbmdQ1KVwvhJNGD8jFJ

im working on this, be patient:)
thanks!

turbochip, 2017-02-12:
> {"associations":{"0":{"robot_id":"MY_ROBOT_ID","deleted":false,"password":":1:1486234474:ONE_PASSOWRD"}},"app_id":"IOS-UUID"}

Where do you get MY_ROBOT_ID from?

rwscott1961, 2017-02-12:
During the discover, about comment 5 above, it is the x's after the Roomba-.

I'm stuck on the IOS-UUID. All my attempts at looking at any of the communication streams end an SSL alert, certificate unknown.

The local MQTT connection I have got past the connect error of 0x01, to a connect error of 0x05. I haven't been able to determine what the app sends for the username and password :(

Thoro, 2017-02-12:
Sooo, for everyone a breakthrough ;)

The iPhone app communicates locally via mqtt (obviously) - it only connects to servers that uses a certificate with the right name (e.g. ROBOTID), maybe you even need a CA, I made my server so that is uses the 1:1 naming like the robot.

Now for the connect packet, it needs to contain the Username and Password Flags, uses the robot id as Identifier and as Username and the password we all got as the password, after sending this to the port at 8883 (via nodejs) you get already pumped with notifications from the robot ;)

```
1040
00044d515454
04
c0
0000
0010 (length hex encoded)
ROBOT_ID (16 bytes)
0010 (length hex encoded)
ROBOT_ID (16 bytes)
0010
PASSWORD (16 bytes)
```

After we got that, now there's a new problem to tackle: the remaining length field sent from the robot is simply wrong:

```
5 ' -- ' <Buffer 30 ef bf bd 01> '30efbfbd01' '0�\u0001'
2 ' -- ' <Buffer 00 08> '0008' '\u0000\b'
8 ' -- ' <Buffer 77 69 66 69 73 74 61 74> '7769666973746174' 'wifistat'
172 ' -- ' <Buffer 7b 22 73 74 61 74 65 22 3a 7b 22 72 65 70 6f 72 74 65 64 22 3a 7b 22 6e 65 74 69 6e 66 6f 22 3a 7b 22 64 68 63 70 22 3a 74 72 75 65 2c 22 61 64 64 72 ... > '7b227374617465223a7b227265706f72746564223a7b226e6574696e666f223a7b2264686370223a747275652c2261646472223a333233323233353732302c226d61736b223a343239343936373034302c226777223a333233323233353532312c22646e7331223a333237333831393431332c22646e7332223a333536393031333532352c226273736964223a2238303a32613a61383a63613a62393a3333222c22736563223a347d7d7d7d' '{"state":{"reported":{"netinfo":{"dhcp":true,"addr":3232235720,"mask":4294967040,"gw":3232235521,"dns1":3273819413,"dns2":3569013525,"bssid":"80:2a:a8:ca:b9:33","sec":4}}}}'
```

That's the Publish (0x30) message sent from the robot, but the length field does not match with the following data, (2 + 8 + 172 = 184), based on the variable length encoding, the robot tells us it's about 3 MB in size, which is not the case. All publish messages  <= 127 bytes work fine.

Sadly the remaining length field makes no sense at all:

30 ef bf bd 01 -> 184 bytes
30 ef bf bd 01 -> 165 bytes
30 ef bf bd 02 -> 288 bytes
30 ef bf bd 01 -> 246 bytes

If anyone can make sense of that, would be great!

rwscott1961, 2017-02-12:
I still get
20 02 00 05
for the CONNACK

You mention that "it only connects to servers that uses a certificate with the right name".
But I am trying to connect to the robot as a client. So how did you create the client key and certificate to get a
20 02 00 00
CONNACK when connecting to the robot??

I am also interested in how you made the key and cert to allow the IOS app to connect. Using the "right name" with anything dealing with PKI is ambiguous, at least to me.

Thoro, 2017-02-12:
1) regarding the CONNACK, you are sending something wrong in the CONNECT message. According to here http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html it means wrong authorization, so I guess username, identifier or password are wrong

2) just a self signed ca + certificate with the same settings for country, state, commonName etc:

id-at-commonName=Roomba-3115C21020313650,id-at-stateOrProvinceName=MA,id-at-localityName=Bedford,id-at-organizationName=iRobot,id-at-countryName=US

Got that info from Wireshark.

3) Not really easy to make you app connect to your PC, I needed a second notebook, configured as Wireless AP, and then rewriting the MAC and the IP address from the iphone to the robot to my local pc - Would not recommend, took me a total of 14 hours (with figuring out, that it won't work on osx :) ) - 0.5 hours if you have the materials and know what your are doing ^^

koalazak, 2017-02-12:
Hi @Thoro i have the same setup as you (2do notebook as AP), cert with asset id and y can see all the trafic mobile<->roomba. But i cant emulate the mobile to connect to the roomba. I get bad auth or bad protocol version response. 
1) Wich lib are using? `mqtt` node module? or plain stl stream to make the packet homemade?
2) You say roomba are using a custom implementation? why mqtt node module doesnt connect?
3) abount the remaining length, the specification says:
`
The Remaining Length is encoded using a variable length encoding scheme which uses a single byte for values up to 127. Larger values are handled as follows. The least significant seven bits of each byte encode the data, and the most significant bit is used to indicate that there are following bytes in the representation. Thus each byte encodes 128 values and a "continuation bit". The maximum number of bytes in the Remaining Length field is four.
`
http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718023

can you share the code you are using to connect and send packages?

thanks a lot!

rwscott1961, 2017-02-12:
Assuming the "username, identifier and password" are the same as the old https ones, I'm pretty sure I have those correct. However, I'm not using a cert with that commonName, stateOrProvinceName, localityName, organizationName, or countryName. Did I mention "right name" was ambiguous when it comes to PKI :) So Im thinking the "wrong authorization" is from the cert.

Not to mention, in some places the password has been base64 encoded, and sometimes not. Details matter :)

As a plus, I'm kinda glad this isn't a piece of cake to figure out. The robot does have access to the entire internal network :)

Thoro, 2017-02-12:
@rwscott1961 the cert is only needed to emilate the robot, not if you want to access it

Pwd and username are the plain ones used everwhere

@koalazak

1) home made plain tls stream, nothing fancy, just implemented mttq according to spec, can share my class tmrw

2) i believe it has a bug, e.g. Regarding 3) - maybe there's also a protocol issue, and you don't need to subscribe, it just constantly publishes stuff

3) yep that's how it should be, but roomba sends shit 😊

rowanmcfarland, 2017-02-12:
I just pulled the certs from my roomba so you can see what it is using, I also have the pem version at the end:

[root@localhost ~]# openssl s_client -showcerts -servername 192.168.1.150 -connect 192.168.1.150:8883 
CONNECTED(00000003)
depth=1 C = US, ST = MA, L = Bedford, O = iRobot, OU = HBU, CN = Roomba CA
verify error:num=19:self signed certificate in certificate chain
verify return:0
---
Certificate chain
 0 s:/C=US/O=iRobot/L=Bedford/ST=MA/CN=Roomba-3107462022006660
   i:/C=US/ST=MA/L=Bedford/O=iRobot/OU=HBU/CN=Roomba CA
-----BEGIN CERTIFICATE-----
MIIDnjCCAoagAwIBAgIIA3bEwTwar2UwDQYJKoZIhvcNAQELBQAwXzELMAkGA1UE
BhMCVVMxCzAJBgNVBAgTAk1BMRAwDgYDVQQHEwdCZWRmb3JkMQ8wDQYDVQQKEwZp
Um9ib3QxDDAKBgNVBAsTA0hCVTESMBAGA1UEAxMJUm9vbWJhIENBMB4XDTE2MDYx
NTAyNDMwOFoXDTI2MDYxNTAyNDMwOFowXzELMAkGA1UEBhMCVVMxDzANBgNVBAoM
BmlSb2JvdDEQMA4GA1UEBwwHQmVkZm9yZDELMAkGA1UECAwCTUExIDAeBgNVBAMM
F1Jvb21iYS0zMTA3NDYyMDIyMDA2NjYwMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8A
MIIBCgKCAQEAo4m/YbtfAF13NyfK3F0olNwgmL3Sj9eSb8ZMvlbtbS/lCRDuSEdj
43L7RzHPUdAOoEf5741JteZYnyC2Z0J8/kALAXcKtXUoPE0StxBGJJ0zzcdNryn/
ByaAHi7pj1G2OpE3Dy6hVddECCCbPm4DrR/xmUEyN8eEidpApUYDXTT3gyWW6VbN
xtAtuuwkOHLjKXRFrbLqN0u3SPui7mz4sqDuT4c1V8zUBLzYSRFJJ9tKoiozsafN
+EBaM73VTNKfpWbTX0rHumQn/CY9/FDJW4ISNf0Fe7nCwR5V/vgpM1D905bkmgwO
WbtqhUzk8UhxrWvSPtBqttOzQPfC8HUosQIDAQABo14wXDAdBgNVHQ4EFgQUJ9WU
9k+xEi6rrLG7Hf7qiQM8dR0wDAYDVR0TAQH/BAIwADAdBgNVHSUEFjAUBggrBgEF
BQcDAQYIKwYBBQUHAwIwDgYDVR0PAQH/BAQDAgWgMA0GCSqGSIb3DQEBCwUAA4IB
AQANVao+f1wLanEZ4zDz/sHQ5j63iRRuKcnrb7RFuNl4Upfx7hcPzCZCTQPi697u
4Bjax+wlkLDFNMZWMtUxXr+DM7k31Pd/O9VOSpVmECxSFn+Lu8v5NoUdke+v1NF4
o1BqsBCcAZEGbWvijPNFgqaVjGAaXnipmFdhgAmDeflxZqKw2jj4+scHV3WjMxTJ
QejvxSg7fuDWOQhm/wfxp5R8g4nMPqJH9vQbo/czdm5UGAn0W6T4dCVpuZI9/NF/
S2F+QcNMMZ3ooIibgL6l8tjBwZ4cKrOPG/rPdPO3XiVZZcdJRgCp8DHUGWn1cCxv
axIy6+GggyKXu+56j5TcktCD
-----END CERTIFICATE-----
 1 s:/C=US/ST=MA/L=Bedford/O=iRobot/OU=HBU/CN=Roomba CA
   i:/C=US/ST=MA/L=Bedford/O=iRobot/OU=HBU/CN=Roomba CA
-----BEGIN CERTIFICATE-----
MIIDSzCCAjOgAwIBAgIBATANBgkqhkiG9w0BAQsFADBfMQswCQYDVQQGEwJVUzEL
MAkGA1UECBMCTUExEDAOBgNVBAcTB0JlZGZvcmQxDzANBgNVBAoTBmlSb2JvdDEM
MAoGA1UECxMDSEJVMRIwEAYDVQQDEwlSb29tYmEgQ0EwHhcNMTUwMzEzMDAwMDAw
WhcNMjUxMjMxMDAwMDAwWjBfMQswCQYDVQQGEwJVUzELMAkGA1UECBMCTUExEDAO
BgNVBAcTB0JlZGZvcmQxDzANBgNVBAoTBmlSb2JvdDEMMAoGA1UECxMDSEJVMRIw
EAYDVQQDEwlSb29tYmEgQ0EwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIB
AQDIooRE+Q2qQ/3SiYbTtwBUwyY/YMncCMrhMoDcvkfJEawL1bHG9eL9c4qXSzec
t22lnY93LsZ80+0nQic5bPz21y89KNAC6Df4Yb4TWUwaZIDocNKjzC4keAKvxmVD
xycKMRM7Vf/f12gVRqXOeFZJUb+RzeIfWGvvNcqSEU234C0REQhtPaT+6jK/Qx04
lepgw8EYM4eoT5Ks10PIGzXemMZu7T44lvCcksog00YAMZaJ7skZ+vnBQhhjQdh+
mJGZtvMCG7Mk1cucu5JPVZwmzT9YKA6NMb3wEsujMI3bhY3QnFtF+a+IrbdjbNy+
/i2mT8yoCs/Q9TaKerT124DDAgMBAAGjEjAQMA4GA1UdDwEB/wQEAwICBDANBgkq
hkiG9w0BAQsFAAOCAQEAWE45Bxe2wpf3/5IRUF0bmf8eXKunsSPlM7+dSo/DKuUz
vyDBQt07NIp9e+OQp3Fojo94C14b27I9UmRPFJRDD3gI8wNKExKCvkUNEAIRVf3w
uRef860RwFodfJWjD4yBEQInLxxCRlCLTJ0gcqiRV8X6HSFCnLGJtIvKqF8hLwdP
m5WfPXr/zHzilTC7745FCZFlOtim5O+nMZMeHZp2urssjFLPvrkb1Q9l+FbF6jMr
gTzcLAd1L3a+NKR/i1TKZ+rn9tNnA9aUAXN9BnxResfBkIEKoQ6HJoFlZyfzKAiW
CRju2zTBnR6vg+kWjPxVCwcGRdzAZeDgc8xFqTxRMw==
-----END CERTIFICATE-----
---
Server certificate
subject=/C=US/O=iRobot/L=Bedford/ST=MA/CN=Roomba-3107462022006660
issuer=/C=US/ST=MA/L=Bedford/O=iRobot/OU=HBU/CN=Roomba CA
---
No client certificate CA names sent
Server Temp Key: DH, 1024 bits
---
SSL handshake has read 2506 bytes and written 499 bytes
---
New, TLSv1/SSLv3, Cipher is DHE-RSA-AES256-SHA256
Server public key is 2048 bit
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : DHE-RSA-AES256-SHA256
    Session-ID: 1B81BDF873C57FD50CB9096DA1FDC8A66EBD6705A0CB0433B9625094B5587978
    Session-ID-ctx: 
    Master-Key: 49D5EF728940DC31FDDDA97E4D1D16CAB58DB6BDBBA8ACDB38B54F0660964B72A72BDA931C2737F83D9E6E1BD28EB13A
    Key-Arg   : None
    Krb5 Principal: None
    PSK identity: None
    PSK identity hint: None
    Start Time: 1486937197
    Timeout   : 300 (sec)
    Verify return code: 19 (self signed certificate in certificate chain)
---
[root@localhost ~]# openssl s_client -showcerts -servername 192.168.1.150 -connect 192.168.1.150:8883 < /dev/null | openssl x509 -outform PEM > roomba.pem
[root@localhost ~]# cat roomba.pem 
-----BEGIN CERTIFICATE-----
MIIDnjCCAoagAwIBAgIIA3bEwTwar2UwDQYJKoZIhvcNAQELBQAwXzELMAkGA1UE
BhMCVVMxCzAJBgNVBAgTAk1BMRAwDgYDVQQHEwdCZWRmb3JkMQ8wDQYDVQQKEwZp
Um9ib3QxDDAKBgNVBAsTA0hCVTESMBAGA1UEAxMJUm9vbWJhIENBMB4XDTE2MDYx
NTAyNDMwOFoXDTI2MDYxNTAyNDMwOFowXzELMAkGA1UEBhMCVVMxDzANBgNVBAoM
BmlSb2JvdDEQMA4GA1UEBwwHQmVkZm9yZDELMAkGA1UECAwCTUExIDAeBgNVBAMM
F1Jvb21iYS0zMTA3NDYyMDIyMDA2NjYwMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8A
MIIBCgKCAQEAo4m/YbtfAF13NyfK3F0olNwgmL3Sj9eSb8ZMvlbtbS/lCRDuSEdj
43L7RzHPUdAOoEf5741JteZYnyC2Z0J8/kALAXcKtXUoPE0StxBGJJ0zzcdNryn/
ByaAHi7pj1G2OpE3Dy6hVddECCCbPm4DrR/xmUEyN8eEidpApUYDXTT3gyWW6VbN
xtAtuuwkOHLjKXRFrbLqN0u3SPui7mz4sqDuT4c1V8zUBLzYSRFJJ9tKoiozsafN
+EBaM73VTNKfpWbTX0rHumQn/CY9/FDJW4ISNf0Fe7nCwR5V/vgpM1D905bkmgwO
WbtqhUzk8UhxrWvSPtBqttOzQPfC8HUosQIDAQABo14wXDAdBgNVHQ4EFgQUJ9WU
9k+xEi6rrLG7Hf7qiQM8dR0wDAYDVR0TAQH/BAIwADAdBgNVHSUEFjAUBggrBgEF
BQcDAQYIKwYBBQUHAwIwDgYDVR0PAQH/BAQDAgWgMA0GCSqGSIb3DQEBCwUAA4IB
AQANVao+f1wLanEZ4zDz/sHQ5j63iRRuKcnrb7RFuNl4Upfx7hcPzCZCTQPi697u
4Bjax+wlkLDFNMZWMtUxXr+DM7k31Pd/O9VOSpVmECxSFn+Lu8v5NoUdke+v1NF4
o1BqsBCcAZEGbWvijPNFgqaVjGAaXnipmFdhgAmDeflxZqKw2jj4+scHV3WjMxTJ
QejvxSg7fuDWOQhm/wfxp5R8g4nMPqJH9vQbo/czdm5UGAn0W6T4dCVpuZI9/NF/
S2F+QcNMMZ3ooIibgL6l8tjBwZ4cKrOPG/rPdPO3XiVZZcdJRgCp8DHUGWn1cCxv
axIy6+GggyKXu+56j5TcktCD
-----END CERTIFICATE-----

rwscott1961, 2017-02-12:
I'm missing something, can't get past the

20 02 00 05

I look forward to the code ....

As for the cloud API, I have to use 1485737522, in place of 1486234474, in the timestamp field to avoid the TooNew or TooOld error, accurate to the second! However I haven't been able to get the app_id correct. If you leave the app_id as a zero length string, you get a nice python stack trace :)

koalazak, 2017-02-12:
The thing with the cert with a `valid` CommonName is that the mobile app is validating the CN field in the roomba cert with the format 'Roomba-{number16}' if it is not valid, then the mobile app disconnect. (the validation is in the mobile app, so you need a selfisgned cert whit this CN if you will perform a sslsplit to snif the trafic.

well, you are getting a response (20 02 00 05), im not :p
can you share the code you are using? @rwscott1961

koalazak, 2017-02-13:
ok, now im sending raw data via tls connection with the content you mention @Thoro:

```javascript
 var bufferAsArray = [
    0x10, 0x40,
    0x00, 0x04, 0x4d, 0x51, 0x54, 0x54,
    0x04,
    0xc0,
    0x00, 0x00,
    0x00, 0x10,
    0x31, 0x32, 0x33, 0x34, 0x35, 0x36, 0x37, 0x38, 0x39, 0x30, 0x31, 0x32, 0x33, 0x34, 0x35, 0x36
    0x00, 0x10,
    0x31, 0x32, 0x33, 0x34, 0x35, 0x36, 0x37, 0x38, 0x39, 0x30, 0x31, 0x32, 0x33, 0x34, 0x35, 0x36
    0x00, 0x10,
    0x31, 0x32, 0x33, 0x34, 0x35, 0x36, 0x37, 0x38, 0x39, 0x30, 0x31, 0x32, 0x33, 0x34, 0x35, 0x36
  ];
  client.write(new Buffer(bufferAsArray));
```

and im getting 0x05 ret code too (`20 02 00 05`)

1) How are you encoding the robot id and password? `new Buffer('1234567890123456')` ? no reverse, no nothing.
2) The password is the same as we have and use in the 1.6.6 firmware? without the mentioned `:1:timestap:pass`? 16bytes length?
3) maybe my robot changes his password with all the MITM stuff I try. Can you try the cmd to get the password? (pressing the home button for some seconds then send the command) and sendme the hex values for the correct commando working?

thanks!

fgomecs, 2017-02-13:
And... new victim here :) 
I will be patient and see how you'll do. Thanks geniuses! I wish I could half 10% of your programming experience.

Thoro, 2017-02-13:
@koalazak

1) just normal string, no mutations
2) yep, 16 bytes, like I wrote, password too
3) could be, whole code attached, just need to enter IP, Username and Password, and you should get loads of output, if not something else is going on -> beta vs. release ... or something like that

Firmware: v2.0.0-34

Btw. using Node v0.12.9 because of my sorrounding frameworks, so adapt where necessary.

[irobot.tar.gz](https://github.com/koalazak/dorita980/files/770297/irobot.tar.gz)

JensBonse, 2017-02-13:
Reminder: As far as I know the password changes everytime you put the roomba in discovery mode.

koalazak, 2017-02-13:
Hi @Thoro, thanks for the code! 
I get ret code 05 unauthorize with your code too :/
I dont kwnow whats going on...
Sniffing I see the password in format `1:1486234474:ONE_PASSOWRD16bytes` 
its weired...

Thoro, 2017-02-13:
That's interesting. 

And when you use that password from sniffing?

koalazak, 2017-02-13:
ret code 05 too

rwscott1961, 2017-02-13:
> Reminder: As far as I know the password changes everytime you put the
> roomba in discovery mode.

If this is true, how does the app re-connect without going through
the setup procedure??

koalazak, 2017-02-13:
@JensBonse I dont think so. (when you link a roomba with other device, the old device stills works for example) And i see the same password in the sniff data.

iosdeveloper, 2017-02-13:
Maybe there is difference between holding only the Wi-Fi button (same password?) and holding both the Wi-Fi and the home button (new password?).

JensBonse, 2017-02-13:
For me the password changed sometime ago. I am not absolutely sure how it happened. I think I changed my home WiFi SSID so I had to reconnect the robot to a new network. It was just an idea, what might have happened.

koalazak, 2017-02-13:
ok, im connected and getting data now :)
you dont believe that: The password is in the `:1:number:pass` format and the number is not a timestamp generated at connection time. Its generated whe you connect the roomba for the first time to a wifi, and continues hardcoded as part of the password every time.
(I try everything (generatng the timestamp) but I never tried a password exaclty as i see in the sniff data hahaha)

@iosdeveloper, @JensBonse the password changes when you connect to other wifi. but not when press home for 2 seconds.

Im going to my `real` work :p today later i will continue with this.

thanks guys!

JensBonse, 2017-02-13:
So I kind of pushed you in the right direction ;)

fgomecs, 2017-02-13:
You both rule! :) 
Go to work... cya later on (I will do the same)

Thoro, 2017-02-13:
For anyone interested, here's the full state reported for $aws/things/MY_ROBOT_ID/shadow/update

```
{ state:
   { reported:
      { country: 'US',
        cloudEnv: 'prod',
        svcEndpoints: { svcDeplId: 'v006' },
        batPct: 9,
        dock: { known: true },
        bin: { present: true, full: false },
        audio: { active: false },
        cleanMissionStatus:
         { cycle: 'quick',
           phase: 'charge',
           expireM: 90,
           rechrgM: 73,
           error: 0,
           notReady: 0,
           mssnM: 88,
           sqft: 386,
           initiator: 'rmtApp',
           nMssn: 60 },
        language: 0,
        noAutoPasses: false,
        noPP: false,
        ecoCharge: false,
        vacHigh: false,
        binPause: false,
        carpetBoost: true,
        openOnly: false,
        twoPass: false,
        schedHold: false,
        lastCommand: { command: 'resume', time: 1487008413, initiator: 'rmtApp' },
        langs:
         [ { 'en-UK': 0 },
           { 'fr-FR': 1 },
           { 'es-ES': 2 },
           { 'it-IT': 3 },
           { 'de-DE': 4 },
           { 'ru-RU': 5 } ],
        bbnav: { aMtrack: 97, nGoodLmrks: 7, aGain: 51, aExpo: 92 },
        bbpanic: { panics: [ 8, 8, 1, 8, 11 ] },
        bbpause: { pauses: [ 2, 0, 0, 13, 0, 6, 16, 0, 2, 2 ] },
        bbmssn:
         { nMssn: 59,
           nMssnOk: 20,
           nMssnC: 32,
           nMssnF: 6,
           aMssnM: 21,
           aCycleM: 22 },
        bbrstinfo: { nNavRst: 9, nMobRst: 0, causes: '0000' },
        cap: { pose: 1, ota: 2, multiPass: 2 },
        sku: 'R960040',
        batteryType: 'li26',
        soundVer: '32',
        uiSwVer: '4582',
        navSwVer: '01.09.09',
        wifiSwVer: '20902',
        mobilityVer: '5309',
        bootloaderVer: '4042',
        umiVer: '5',
        softwareVer: 'v2.0.0-34',
        tz:
         { events: [ { dt: 0, off: 60 }, { dt: 0, off: 120 }, { dt: 0, off: 0 } ],
           ver: 2 },
        timezone: 'Europe/Vienna',
        name: 'Staubi',
        cleanSchedule:
         { cycle: [ 'none', 'start', 'start', 'start', 'start', 'start', 'none' ],
           h: [ 9, 9, 9, 9, 9, 9, 0 ],
           m: [ 0, 0, 0, 0, 0, 0, 0 ] },
        bbchg3:
         { avgMin: 435,
           hOnDock: 944,
           nAvail: 78,
           estCap: 7451,
           nLithChrg: 24,
           nNimhChrg: 0,
           nDocks: 22 },
        bbchg: { nChgOk: 22, nLithF: 0, aborts: [ 0, 0, 0 ] },
        bbswitch: { nBumper: 21981, nClean: 84, nSpot: 11, nDock: 22, nDrops: 146 },
        bbrun:
         { hr: 27,
           min: 52,
           sqft: 109,
           nStuck: 13,
           nScrubs: 8,
           nPicks: 160,
           nPanics: 56,
           nCliffsF: 308,
           nCliffsR: 642,
           nMBStll: 5,
           nWStll: 0,
           nCBump: 0 },
        bbsys: { hr: 1006, min: 54 } } } }
```

rwscott1961, 2017-02-13:
> ok, im connected and getting data now :)
> you dont believe that: The password is in the :1:number:pass format
> and the number is not a timestamp generated at connection time. Its
> generated whe you [...]
When you say the password changes, do you mean the "number" part, the
"pass" part, or both?

rwscott1961, 2017-02-13:
I'm guessing that this timestamp is not the same as the one used for the
cloud communications.

I have still not managed to get a successful sslsnif to work. in tcp
mode it connects right up, but all of the logs are encrypted. In ssl, or
https, mode I get:

Connecting to [192.168.1.102]:8883
===> Original server certificate:
Subject DN: /C=US/O=iRobot/L=Bedford/ST=MA/CN=Roomba-3107462000221890
Common Names: Roomba-3107462000221890
Fingerprint: A3:2C:D7:7D:B4:46:3A:0B:69:7B7C:FD:AC:1D:A2:F7:8B:D4:85:67
Certificate cache: MISS
===> Forged server certificate:
Subject DN: /C=US/O=iRobot/L=Bedford/ST=MA/CN=Roomba-3107462000221890
Common Names: Roomba-3107462000221890
Fingerprint: 9D:4A:FC:5D:06:17:A4:9C:25:6D82:F6:6F:0B:CA:43:A0:71:BD:8A
SSL connected to [192.168.1.102]:8883 TLSv1.2 DHE-RSA-AES256-SHA256
Received privsep req type 01 sz 109 on srvsock 9
Error from bufferevent: 0:- 336151574:1046:sslv3 alert certificate
unknown:20:SSL routines:148:SSL3_READ_BYTES
SSL disconnected to [192.168.1.102]:8883
SSL disconnected from [192.168.1.101]:52849

Has anyone done a successful sniff with the android app??

koalazak, 2017-02-14:
Hi @rwscott1961 Im not sure but I think both change.
I use iOS :/

koalazak, 2017-02-14:
Hi everyone!
dorita980 v3.0 released! and support fireware 2.0.0 for local mqtt API.
I preserve the old methods and add others taking advantage of the new mqtt API.
Also add some events to handle all the new data.

To upgrade, change your package.json to use `dorita980: "^3.0.0"` then `npm update` in your project.

TODO:
- getPassword
- Cloud API

I will close this issue and open a new one for getPassword in firmware 2.0.0 and other for Cloud API

thanks for everybody!

[66 more replies]

closed 2017-02-16 by koalazak