koalazak/dorita980 #12: Error package.json

Melkor82, 2017-01-19:
I tried today to install and test this package and got the error 

npm ERR! path /home/pi/package.json

so i open the npm-debug.log as suggested

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/nodejs',
1 verbose cli   '/usr/bin/npm',
1 verbose cli   'run',
1 verbose cli   'getpassword',
1 verbose cli   '192.168.1.52' ]
2 info using npm@4.0.5
3 info using node@v7.4.0
4 verbose stack Error: ENOENT: no such file or directory, open '/home/pi/packag$
5 verbose cwd /home/pi
6 error Linux 4.4.0-1-rpi2
7 error argv "/usr/bin/nodejs" "/usr/bin/npm" "run" "getpassword" "192.168.1.XXX"
8 error node v7.4.0
9 error npm  v4.0.5
10 error path /home/pi/package.json
11 error code ENOENT
12 error errno -2
13 error syscall open
14 error enoent ENOENT: no such file or directory, open '/home/pi/package.json'

koalazak, 2017-01-19:
Can you please share the complete steps you are doing to install and use the library?

closed 2017-01-22 by koalazak

Melkor82, 2017-01-24:
i installed nodejs with version 7.4 and i got npm with version 4.0.5. The thing is that i am running a custom version of debian called openhabian. My final goal was to connect roomba with openhab through rest api

https://github.com/openhab/openhabian

koalazak, 2017-01-24:
ok, but your are following the readme steps to use dorita980?
Can you please share the complete steps you are doing to install and use dorita980?

Melkor82, 2017-01-24:
yes, i followed it. I first installed nodejs with sudo apt-get install nodejs and got version 7.4.0
then i installed npm and got version 4.0.5. Then i installed dorita 980 as the readme steps with `npm install dorita980 --save`. then i tried to recover my roomba password with the command `npm run getpassword 192.168.1.XX` and got the package.json error

koalazak, 2017-01-24:
Ok, that is the problem. To get your password the readme says (note the clone):
```
git clone https://github.com/koalazak/dorita980.git
$ cd dorita980
$ npm install
$ npm run getpassword
```
note in what you are doing: you are installing dorita980 to use in your own project creating an empy package.json and when you run `npm run getpassword` you are calling a script in your own empty package not inside dorita980 source code.
There are two diferent things.
After you get the password using this metodo (with clone) you are able to install dorita98o in your own project and run the examples codes.

Melkor82, 2017-01-24:
thanks it worked! Sorry for the loss of time

airjesse123, 2017-12-06:
Alright I'm feeling like a bit of an idiot. I am having the same issue as Melkor82 even though I've cloned the repo. 
I am in the directory /home/pi/Documents/dorita980 and have already run npm install.
When I run `
npm run getpassword 192.168.1.22`

I get
```
npm ERR! Error: ENOENT: no such file or directory, open '/home/pi/Documents/dorita980/node_modules/getpassword/package.json'
npm ERR!     at Error (native)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.9.59-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "run" "getpassword" "192.168.1.22"
npm ERR! cwd /home/pi/Documents/dorita980
npm ERR! node -v v4.8.2
npm ERR! npm -v 1.4.21
npm ERR! path /home/pi/Documents/dorita980/node_modules/getpassword/package.json
npm ERR! syscall open
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/Documents/dorita980/npm-debug.log
npm ERR! not ok code 0
```