koalazak/dorita980 #19: Error running getpassword

turboc1208, 2017-02-12:
Let me preface this with I know absolutely nothing about npm or nodejs.  I am trying to get my userid and password from my roomba.  However I am getting the following error.

```
pi@hass2:~/code/roomba/dorita980$ npm run getpassword 192.168.2.161
npm ERR! Error: ENOENT, open '/home/pi/code/roomba/dorita980/node_modules/getpassword/package.json'
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.4.38-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "run" "getpassword" "192.168.2.161"
npm ERR! cwd /home/pi/code/roomba/dorita980
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! path /home/pi/code/roomba/dorita980/node_modules/getpassword/package.json
npm ERR! code ENOENT
npm ERR! errno 34
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/code/roomba/dorita980/npm-debug.log
npm ERR! not ok code 0
pi@hass2:~/code/roomba/dorita980$ 

```
I think I followed the instructions from the readme, but as you can see, it's not getting me anywhere.  I tlooked like it was complaiing that the package.json file was not found, so I created the directory structure (node_modules was there, but the password directory under that was not).  When I created that structure and put the package.json file that was in the dorita980 directory in there, the app seemed to run, but no output at all.  It just came back to the prompt.  I did put the robot into what I call discovery mode by pressing the house until it beeped before each attempt.  Please help.

koalazak, 2017-02-12:
I think you are not following the readme steps properly:
see https://github.com/koalazak/dorita980/issues/12#issuecomment-274808204

(btw, check if you have 1.6.6 firmware #10)

closed 2017-02-12 by koalazak

turboc1208, 2017-02-12:
Well the firmware thing kind of puts a stop to what I wanted to do for now.  L

Thanks

 

see #12 (comment)