koalazak/dorita980 #21: getPassword problems

ftkalcevic, 2017-02-16:
I'm having problems getting the password.  

When I followed the instructions (using nodejs 7.4 - there are runtime errors using 7.5) I get...
```

$ npm run getpassword 10.0.0.210

> dorita980@3.0.2 getpassword C:\dorita980
> node ./bin/getpassword.js "10.0.0.210"

Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...

Looking for robots...
Robot found! with blid/username: XXXXXXXXXX
{ ver: '2',
  hostname: 'Roomba-XXXXXXXXXX',
  robotname: 'Bob',
  ip: '10.0.0.210',
  mac: '11:22:33:44:55:66',
  sw: 'v2.0.0-34',
  sku: 'R980000',
  nc: 0,
  proto: 'mqtt' }
data=> 00000000: efbf bd05                                o?=.
 <=
Error getting password. Follow the instructions and try again.
```

I added the "data=>" bit - it's hexy dumping the packet that came back.

Any ideas?

koalazak, 2017-02-16:
Sounds like you are sending the packet before that pressing the HOME button.
First press the HOME button and when light blinks press any key in the script.

IngmarStein, 2017-02-16:
Same here. I made sure to let the script only progress when the WiFi light is blinking:
```
$ npm run getpassword 192.168.178.22

> dorita980@3.0.4 getpassword …/dorita980
> node ./bin/getpassword.js "192.168.178.22"

Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Looking for robots...
Robot found! with blid/username: xxxxxxxxxxxxxxxxxx
{ ver: '2',
  hostname: 'Roomba-xxxxxxxxxx',
  robotname: 'Robbie',
  ip: '192.168.178.22',
  mac: '80:A5:89:48:14:08',
  sw: 'v2.0.0-34',
  sku: 'R98----',
  nc: 0,
  proto: 'mqtt' }
Error getting password. Follow the instructions and try again.
```

koalazak, 2017-02-16:
mm maybe a problem whit the node 7 Buffer function? you are usgin nvm? can you try with node 6?

marcodesantis71, 2017-02-16:
HI
I have the same problem.

I use node version 6.9.5

koalazak, 2017-02-16:
today at night I will try in node 6. All my tests was in node 5

ftkalcevic, 2017-02-16:
Thank you.  I had to go back to node version 5.12.0 to get it to work.

koalazak, 2017-02-16:
nice! 
There are breaking changes in how Buffer() works in node > 5
Today I will fix that for node 6 and 7

thanks you

koalazak, 2017-02-16:
fixed in a very ugly way :p
upgrade to 3.0.5

thanks!

closed 2017-02-16 by koalazak

barqers, 2018-08-11:
Hello,
I'm having the same issue, can you confirm it's still working?

Thank you...

koalazak, 2018-08-11:
Yes, it still works. re-tested with the latest roomba v2.4.6-3 firmware and node.js 5, 6, 7 and 8.

barqers, 2018-08-11:
Just realized you had to hold the home button until it beeps not until it lights up. Oh boy, after installing it via pip, python, python without running setup, and finally through homebrew on my mac, finally got it working.

Awesome script and thank you for posting this.

wiuwiu, 2019-01-30:
@koalazak I have tried couple of times to get the password without any luck. I used Raspbian on Raspberry Pi and Ubuntu 18.04 LTS on my laptop. I used nodejs 8.15 and 10.15.
Everytime I get `Error getting password` message.
My firmware is v2.4.6-3.
What else should I try?

LukaszKoc, 2021-11-12:
instructions to make it work
1 hold home + spot to get info configuration mode
2 get the robot ip np https://www.pcwdld.com/how-to-scan-network-for-ip-addresses
run script with ip from step 2 
boom it works