koalazak/dorita980 #22: 2 Robots (980) same network

steedferns, 2017-02-19:
I created a homebridge (homekit) plugin. It works fine for 1 robot at a time. The second robot is a hit/miss. Anyone know why this would happen?

koalazak, 2017-02-20:
I dont have 2 robots to try and I dont kwno how homebridge works
no idea :/

koalazak, 2017-02-20:
If you use the dorita980 example snippet doplicated to send commands to 2 robots. that works?

koalazak, 2017-02-20:
can you share that code?

steedferns, 2017-02-20:
I run the code below twice for each robot.  (Not in the same program!) 

The both return ok, but the second does not start.

`
var dorita980 = require('dorita980');

var myRobotViaLocal = new dorita980.Local('id', 'password', 'ip'); // robot IP address

myRobotViaLocal.on('connect', function () {
// start!
myRobotViaLocal.start().then(() => {
myRobotViaLocal.end(); // disconnect to leave free the channel for the mobile app.
}).catch((err) => {
console.log(err);
});
});
`

JensBonse, 2017-02-20:
Are both robots on firmware 2.0.0-34?
Sounds like the old cloud api. It still answers like expected but does not trigger the robot.

steedferns, 2017-02-20:
Yes, both are the same firmware.  I can only get robot 2 to work if the other is "inactive/sleeping"? Its a hit or miss

koalazak, 2017-02-20:
maybe the robots talk with each others when an action is performed? and the channel is occuped.
Have you the mobile app offline?
Can you send commands at the same time with 2 mobile apps to the robots?

steedferns, 2017-02-20:
Yes, I can send commands at the same time using the app.

koalazak, 2017-02-22:
news?

koalazak, 2017-03-06:
reopen if needed

closed 2017-03-06 by koalazak

blaster2014, 2017-11-20:
Have two 980 on the same network and when both are running at the same time they both don't complete the job.  One will work for an hour and return back to base as complete.  In fact only 1/3 of the home has been completed.