koalazak/dorita980 #24: Documentation "How to get your username/blid and password" does not match with actual output

Lul4n, 2017-02-20:
When using the getpassword js in order to detect the blid/username and password for my roomba 966, i only get the password.

When taking a look into the code of getpassword.js it only logs the password
`function checkV2 () {
      ...
      console.log('Password=> ' + new Buffer(data).slice(sliceFrom).toString() + ' <= Yes, all this string.');
      console.log('Use this credentials in dorita980 lib :)');
      ...
   }`

Am i missing something important?

koalazak, 2017-02-20:
The blid is printed before the password by the discovery() method. Maybe the UDP brodcast messages are not allowed in your network? or maybe the romba966 has other discovery method?
(discovery() sends a packet to brodcast and roombas in the network response with his information)

I dont have a roomba966 to try how is his discovery method :/

jorisal, 2017-02-21:
Hi, the same happens to me with a 980 / firmware 2.0.0-34
I'll check UDP broadcast messages on my network... may be is this.

koalazak, 2017-02-21:
what version of node.js you have?

jorisal, 2017-02-21:
[user@host ~]$ node -v
v6.9.5

npm.x86_64 : Node.js Package Manager

[user@host ~]$ rpm -qa | grep -i npm

npm-3.10.10-1.6.9.5.1.fc25.x86_64

[user@host ~]$ uname -a

Linux ed1624 4.9.9-200.fc25.x86_64 #1 SMP Thu Feb 9 17:28:13 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

[user@host ~]$ cat /etc/redhat-release 

Fedora release 25 (Twenty Five)

steedferns, 2017-02-21:
I have the same problem. Luckily I had my BLID from before

koalazak, 2017-02-21:
Check and allow UDP broadcast messages on your network.

And then run:

```
var dorita980 = require('dorita980');

dorita980.discovery(function (ierr, data) {
  console.log('err', ierr);
  console.log(data);
});
```

avaneeshd, 2017-02-22:
I'm facing the same issue and I dont know how to enable UDP broadcast on my network. Will it work if we send UDP to the known roomba IP instead of broadcast?

avaneeshd, 2017-02-22:
This worked for me: 
In discovery.js, line: 29
`server.bind(5678, function () {
    const message = new Buffer('irobotmcs');
    server.send(message, 0, message.length, 5678, '<YOUR_ROOMBA_IP>');
  });
}`

removed server.setBroadcast(true); and in server.send() instead of broadcast IP add your roomba IP. It returns the BLID and other details.

jorisal, 2017-02-22:
That's the trick!

[user@host dorita980]$ grep -i "server.bind" -A 4 lib/discovery.js
  server.bind(5678, function () {
    const message = new Buffer('irobotmcs');
    server.setBroadcast(true);
    server.send(message, 0, message.length, 5678, '255.255.255.255');
  });

[user@host dorita980]$ cp lib/discovery.js lib/discovery.js.bak

[user@host dorita980]$ sed -i 's/server.setBroadcast(true);/\/*server.setBroadcast(true);*\//g' lib/discovery.js

[user@host dorita980]$ sed -i 's/255.255.255.255/192.168.1.2/g' lib/discovery.js

[user@host dorita980]$ grep -i "server.bind" -A 4 lib/discovery.js
  server.bind(5678, function () {
    const message = new Buffer('irobotmcs');
    /*server.setBroadcast(true);*/
    server.send(message, 0, message.length, 5678, '192.168.1.2');
  });

[user@host dorita980]$ npm run getpassword 192.168.1.2

> dorita980@3.0.6 getpassword /home/user/dorita980
> node ./bin/getpassword.js "192.168.1.2"

Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Looking for robots...
Robot found! with blid/username: XXXXXXXXXXXXXXXX
{ ver: '2',
  hostname: 'Roomba-XXXXXXXXXXXXXXXX',
  robotname: 'R980',
  ip: '192.168.1.2',
  mac: 'XX:XX:XX:XX:XX:XX',
  sw: 'v2.0.0-34',
  sku: 'R980040',
  nc: 0,
  proto: 'mqtt' }
Password=> XXXXXXXXXXXXXXXX <= Yes, all this string.
Use this credentials in dorita980 lib :)

jorisal, 2017-02-22:
Now happens this...

[user@host rest980]$ cat config/default.json
{
  "port": 3000,
  "blid": "XXXXXXXXXXXXXXXX",
  "password": "XXXXXXXXXXXXXXXX",
  "robotIP": "192.168.1.2",
  "basicAuthUser": null,
  "basicAuthPass": null,
  "sslKeyFile": null,
  "sslCertFile": null
}

[user@host rest980]$ DEBUG=rest980:* npm start

> rest980@1.5.3 start /home/user/rest980
> node ./bin/www

  rest980:server Listening on port 3000 +0ms

[user@host rest980]$ curl http://localhost:3000/api/local/info/mission

GET /api/local/info/mission 500 53.028 ms - 1336
RequestError: Error: connect ECONNREFUSED 192.168.1.2:443
    at new RequestError (/home/user/rest980/node_modules/request-promise-core/lib/errors.js:14:15)
    at Request.plumbing.callback (/home/user/rest980/node_modules/request-promise-core/lib/plumbing.js:87:29)
    at Request.RP$callback [as _callback] (/home/user/rest980/node_modules/request-promise-core/lib/plumbing.js:46:31)
    at self.callback (/home/user/rest980/node_modules/request/request.js:186:22)
    at emitOne (events.js:96:13)
    at Request.emit (events.js:188:7)
    at Request.onRequestError (/home/user/rest980/node_modules/request/request.js:845:8)
    at emitOne (events.js:96:13)
    at ClientRequest.emit (events.js:188:7)
    at TLSSocket.socketErrorListener (_http_client.js:310:9)
    at emitOne (events.js:96:13)
    at TLSSocket.emit (events.js:188:7)
    at emitErrorNT (net.js:1278:8)
    at _combinedTickCallback (internal/process/next_tick.js:74:11)
    at process._tickCallback (internal/process/next_tick.js:98:9)

Maybe i forget something?

koalazak, 2017-02-22:
Ok, that confirm you dont allow UDP brodcast in your network.
I will get adventage about getpassword require the IP and I will send the packet directly instead of use brodcast.

thanks guys!

@jisnardo rest980 its not updated yet: https://github.com/koalazak/rest980/issues/14

closed 2017-02-22 by koalazak