koalazak/dorita980 #27: can't get password

alwaysbadboy, 2017-02-25:
Hi,
It shows error on getting password, any idea ?
looks like data.length is 0 
I m running on ubuntu 16.10 desktop 
node -v 4.2.6

Thanks 

-------------------
Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Robot Data:
{ ver: '2',
  hostname: 'Roomba-xxxxxxxx',
  robotname: 'Roomba980',
  ip: '192.168.1.123',
  mac: 'xx:xx:xx:xx:xx:xx',
  sw: 'v2.0.0-34',
  sku: 'R98----',
  nc: 0,
  proto: 'mqtt',
  blid: 'xxxxxxx' }
Error getting password. Follow the instructions and try again.

koalazak, 2017-02-25:
upgrade your node to latest 4 branch (4.7.3)

closed 2017-02-25 by koalazak

alwaysbadboy, 2017-02-25:
thank you, it works after i upgrade to latest version