koalazak/dorita980 #30: Port 8883 not opened?

kavod, 2017-03-04:
Hi,

I will turn crazy with my update 2.0.0-34.
When I use discovery function, I correctly find my Roomba 980
```
~/dorita980$ node discover.js 
Looking for robots...
Robot found! with blid/username: 3115800850720780
{ ver: '2',
  hostname: 'Roomba-3115800850720780',
  robotname: 'Roomba',
  ip: '192.168.0.18',
  mac: '74:C6:3B:96:FF:91',
  sw: 'v2.0.0-34',
  sku: 'R980040',
  nc: 1,
  proto: 'mqtt' }
{ ver: '2',
  hostname: 'Roomba-3115800850720780',
  robotname: 'Roomba',
  ip: '192.168.0.18',
  mac: '74:C6:3B:96:FF:91',
  sw: 'v2.0.0-34',
  sku: 'R980040',
  nc: 1,
  proto: 'mqtt' }
```

But, the getpassword function return me errors even if I follow instructions (and WIFI led is blinking)
```
~/dorita980$ npm run getpassword 192.168.0.18

> dorita980@3.0.8 getpassword /home/boris/dorita980
> node ./bin/getpassword.js "192.168.0.18"

Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
events.js:154
      throw er; // Unhandled 'error' event
      ^

Error: connect ECONNREFUSED 192.168.0.18:8883
    at Object.exports._errnoException (util.js:893:11)
    at exports._exceptionWithHostPort (util.js:916:20)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1075:14)

npm ERR! Linux 4.4.0-64-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "run" "getpassword" "192.168.0.18"
npm ERR! node v5.12.0
npm ERR! npm  v3.8.6
npm ERR! code ELIFECYCLE
npm ERR! dorita980@3.0.8 getpassword: `node ./bin/getpassword.js "192.168.0.18"`
npm ERR! Exit status 1
```

Actually, I believe that the 8883 port is closed
```
~/dorita980$ netcat -n -v 192.168.0.18 8883
netcat: connect to 192.168.0.18 port 8883 (tcp) failed: Connection refused
```
Do I miss something obvious? Any help would be appreciate.

koalazak, 2017-03-05:
sounds like you are not in the same network or a firewall in the middle.

kavod, 2017-03-05:
After hours of tests/resets/Wifi configurations, it seems the issue was due to... dirty pins on dock station :/

Actually, after full cleaning, the port is accessible again.
Not sure at 100% of this cause... but the password recovery process is no OK for me.

Thanks for your answer!

closed 2017-03-05 by kavod

kavod, 2017-03-05:
Actually, it seems simpler than my nebulous first explanation. I now quite sure it is because of the iRobot app which was running during my tests.
So, if someone has the same issue, be care to switch off the app on your mobile before use the dorita980 lib.
Regards

Anomen, 2021-11-07:
Thank you @jdog09! Same issue here: I disconnected Home Assistant and all of sudden the script worked well!