koalazak/dorita980 #31: new firmware v2.2.5-2

koalazak, 2017-03-05:
getpassword works, state events and cleaning preferes too.
commands like start/stop/dock doesnt work :/
next week i will try to fix

soundVer: '31'
uiSwVer: '4582'
navSwVer: '01.09.09' >> '01.11.02'
wifiSwVer: '20902' >> '20923'
mobilityVer: '5309' >> '5420'
bootloaderVer: '3580',
umiVer: '5' >> '6'
softwareVer: 'v2.0.0-34' >> 'v2.2.5-2'
svcEndpoints.svcDeplId: 'v005' >> 'v007'

and a new creepy property in the state: mapUploadAllowed: true
disable with: myRobotViaLocal.setPreferences({mapUploadAllowed: false}) :trollface:

iosdeveloper, 2017-03-06:
Start/stop/dock work for me in v2.2.5-2. ¯\_(ツ)_/¯

koalazak, 2017-03-06:
its true, works for me now ¯_(ツ)_/¯

letier, 2017-03-06:
That's great news. My roomba didn't upgrade, yet.

koalazak, 2017-03-06:
I will update the README to encorage poeple to block their roomba internet connection if enjoy dorita980 to avoid the updates until know if dorita980 still works...

closed 2017-03-06 by koalazak

Mandy2000, 2017-03-24:
well, for me it works SOMETIMES ... tried it TODAY with my Roomba 960 ... works in about 50% of the cases only

BladeZX9123, 2017-03-26:
Is there a way to get back the history from the app of the successful runs?

koalazak, 2017-03-26:
the history is gone cos you block the internet connection and the robot cant send the information.
I dont know if allowing the internet connection now, the robot send the old history. I dont think so.

The final solution is blocking only the urls used to update the firmware and allow the rest. ( I dont know wich they are)