koalazak/dorita980 #32: No blid/username output!?

sailorsail, 2017-03-06:
In the README.dm it stats that running
`$ npm run getpassword <robotIP>`
one gets that blid/username:
> Looking for robots...
> Robot found! with blid/username: xxxxxxxxxxxxx

However in the output I've got the blid/username line is missing:
> Maresia:dorita980 rmnm$ npm run getpassword 192.168.1.66
> 
> > dorita980@3.0.8 getpassword /Users/someUser/dorita980
> > node ./bin/getpassword.js "192.168.1.66"
> 
> Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
> Then press any key here...
> Robot Data:
> { ver: '2',
>   hostname: 'Roomba-6943C91C12422640',
>   robotname: 'Raft',
>   ip: '192.168.1.66',
>   mac: '80:A5:89:60:D8:48',
>   sw: 'v2.0.0-34',
>   sku: 'R98----',
>   nc: 0,
>   proto: 'mqtt',
>   blid: '6943C91C12422640' }
> Password=> **************** <= Yes, all this string.
> Use this credentials in dorita980 lib :)

Looking to `dorita980.getRobotIP()` maybe I can get hold of the blid/username, is that right?

sailorsail, 2017-03-06:
Sorry. It is actually there in the line:
> blid: '6943C91C12422640' }

It is just the output in the README.md that is misleading!

koalazak, 2017-03-06:
oh, is the old output for firmware 1.
fixed.

closed 2017-03-06 by koalazak