koalazak/dorita980 #35: My Roomba 980 can't automatic update firmware 2.0

saclo, 2017-03-10:
Hello
My Roomba 980 still firmware 1.6.6 . Who can help me how to  update to firmware 2.0 . i'm waiting about 3 month and don't see anything .
Thank you

koalazak, 2017-03-10:
Are you sure your roomba has internet connection?
To check that verify if you see the recent history, or if you can control your robot with the official app when you are not in home.

If the robot has internet connection OK, try a soft reset: (hold CLEAN button for 10 seconds until the sound). And wait a few days more.

closed 2017-03-10 by koalazak

saclo, 2017-03-10:
i sure , my roomba has internet connection . i have reset many time but nothing .
Please help me .

Anderman, 2017-04-16:
@saclo Did you solve the problem. My robot 980 is stil on 1.6.4