koalazak/dorita980 #36: V2.2.5-2 looks like they broke it again

sheppy99, 2017-03-12:
My Roomba has had its firmware upgraded again and dorita980 seems broken. Also the app seems to only connect via the net now, running node to query it just gives an empty reply, and ports 1883 and 1883 appear closed.
Knew I should have bought a different make with an open APi!!

koalazak, 2017-03-13:
According to https://github.com/koalazak/dorita980/issues/31 dorita980 stills works.
If you has the official app open, then dorita980 cant connecto to the robot. Close all other conections, soft reset the robot (10 seconds CLEAN button) and try again.

sheppy99, 2017-03-13:
Is this locally or via the cloud?

koalazak, 2017-03-13:
locally. Cloud api does not work yet with 2.x.x.

sheppy99, 2017-03-13:
That worked - what's the easiest way of stopping these clowns from updating it again?

koalazak, 2017-03-14:
If you block the internet connectio to your robot then, the history will not be available and you can not control the robot from outside your home.

closed 2017-03-14 by koalazak

sheppy99, 2017-03-14:
Given that with the new firmware. the robot seems to shutdown its local comms the moment the app has connected, the apps not really usable now anyway if I want to use the APi with my home automation, so nothing more is lost with blocking it. 
I much preferred it before they started meddling!