koalazak/dorita980 #37: App changes for the better and its still working

sheppy99, 2017-03-23:
My app has recently been updated to V2.0.0 and now it has a cleaning map and push notifications. It also connects via the LAN again. Switching to another app on iOS allows Dorita980 to work again too, so the single LAN user restriction remains, although it does allow more than 1 ip address to connect.
Is it worth starting a firmware version thread so we can easily see when things are updated and if it still works? Currently I've blocked it from the internet so iRobot can't break things.
I'm on 2.2.5-2 dated 6th March 2017 with App V2.0.0 dated 14th March 2017, and all seems well

koalazak, 2017-03-26:
Its a good idea. maybe a table in the readme.md with all the features by firmware and if its implemented in dorita980 or not and the status of progress. what you think?

sheppy99, 2017-03-26:
I think that would be a great idea, assuming it doesn't become a pain to keep up to date.

koalazak, 2017-05-22:
done.

closed 2017-05-22 by koalazak