koalazak/dorita980 #39: Complete API Support

deckar01, 2017-03-23:
Here is a list of commands and preferences used in the iRobot Home v2.0.0 android app:

I checked off the ones that are already implemented.

That debug preference looks interesting. 😈 

### Commands

 - [ ] `BYE`
 - [ ] `CLEAN`
 - [ ] `DLPKG`
 - [x] `DOCK`
 - [ ] `FBEEP`
 - [ ] `FIND`
 - [ ] `IPDONE`
 - [ ] `OFF`
 - [ ] `PATCH`
 - [x] `PAUSE`
 - [ ] `PROVDONE`
 - [ ] `QUICK`
 - [ ] `RECHRG`
 - [ ] `RESET`
 - [x] `RESUME`
 - [ ] `SLEEP`
 - [ ] `SPOT`
 - [ ] `SPRAY`
 - [x] `START`
 - [ ] `STARTVIBRATE`
 - [x] `STOP`
 - [ ] `STOPVIBRATE`
 - [ ] `WAKE`
 - [ ] `WIFISCAN`
 - [ ] `WIPE`
 - [ ] `WLAPOFF`
 - [ ] `WLAPON`
 - [ ] `WLLOGFLUSH`
 - [ ] `WLSTOFF`
 - [ ] `WLSTON`

### Preferences

 - [ ] `autoEvacCount`
 - [ ] `autoEvacFlags`
 - [ ] `autoEvacModel`
 - [ ] `bbrun`
 - [x] `binPause`
 - [x] `carpetBoost`
 - [x] `cleanSchedule`
 - [ ] `country`
 - [ ] `debug`
 - [ ] `lastSwUpdate`
 - [ ] `manualUpdate`
 - [ ] `method`
 - [ ] `milestones`
 - [ ] `mission`
 - [x] `openOnly`
 - [ ] `postalCode`
 - [ ] `preventativeMaintenance`
 - [ ] `regDate`
 - [ ] `robotLanguage`
 - [ ] `robotName`
 - [ ] `robot_status`
 - [ ] `schedHold`
 - [ ] `sku`
 - [ ] `softwareVersion`
 - [ ] `status`
 - [ ] `swUpdateAvailable`
 - [x] `twoPass`
 - [ ] `tzName`
 - [ ] `value`
 - [ ] `wifiDiagnostics`

koalazak, 2017-03-26:
I see this commands reversing the android app and I just implement the most useful.
Maybe is a good idea expose a new function like `myRobot.sendRawCommand(json)` to experiment with all the not implementend commands and preferences.
I will do that.

koalazak, 2017-03-26:
updated documentation with `.publish()` method. thanks! #

closed 2017-03-26 by koalazak

deckar01, 2017-03-27:
Awesome! Thank you.

drinfernoo, 2021-06-03:
What does `debug` end up doing? I'm attempting to find any way to get the "cleaned area" from the response... It seems the `sqft` field in mission status is no longer updated as of firmware 2.4.16-126 (or thereabouts).