koalazak/dorita980 #40: dorita980/rest980 worked sometimes and now all seems broken

Mandy2000, 2017-03-25:
Hi folks,
I bought a 960 yesterday. It is on v2.2.5-2 and yesterday I was able to get my blid/pass via dorita980@3.0.10 without any problems. When using the rest API i realized that even the browser ALWAYS said "OK" the robot only started in about 50% of the cases. But it DID start sometimes.
In the evening I updated node from v6.1.0 to v.6.10.1. Later that day I realized that commands to the Roomba were not working anymore. When sending a command I always got the response that "Safari can't open the page 192.168.xxx.yyy:3000/api/local/action/start because it can't connect to server 192.168.xxx.yyy" even the server definitely is running.

Today I also realized, that the 960 isn't showing the cleaning map on the original Roomba App after cleaning although it did perfectly yesterday on its first attempt.

Today I tried dority980 again to tell me my password and now I get an error that "Failed at the dorita980@3.0.10 getpassword script 'node ./bin/getpassword.js "192.168.xxx.yyy"'."

Is it possible that iRobot again changed something yesterday evening or is it maybe related to my node/npm update? I actually have node 6.10.1 and npm 3.10.10 installed. I also tried to "downgrade to node 6.1.0 again but this did NOT solve any problem ...

I am really confused because presently I do not know where to start searching for the cause ...

Thanks for your support
Mandy

JensBonse, 2017-03-25:
I got a 980 and everything works fine for me.

Regarding rest980: How did you make sure that the server is still running? If Safari tell you that it cannot connect to the host, than normally the web server is somehow down.

Regarding the clean map: I saw this issue once. The reason was, that I was too fast with opening the view. After I retried it, it worked perfectly fine.

Regarding getPassword: Did you follow the instructions? Perhaps your iRobot App was running?

Mandy2000, 2017-03-25:
Well, according terminal window there were still entries when I interrupted. 

GET /api/local/info/mission 200 2731.603 ms - 250
GET /api/local/action/start 304 2292.965 ms - -
GET /api/local/action/start 200 2305.150 ms - 11

By the way can someone please explain what I can read out of the log? Sometimes there is a "200", sometimes a "304" and after the milliseconds there is sometimes a number (here e.g. 250 or 11)

Nevertheless now it seems to work again (I didn't change anything). The clean map is definitely not available. But I tried two further cleaning processes and there the map is shown correctly again. So I think there was really a "problem" on the iRobot backend maybe.

What still is NOT working to my satisfaction is that the start signal (http://192.168.xxx.yyy:3000/api/local/action/start) very very often does NOTHING to the Roomba even the Browser answers with {"OK":null} and the Terminal answers with "GET /api/local/action/start 304 2316.026 ms - -"

thanks again for help

koalazak, 2017-03-26:
You have the official mobile app open? or in background?
Some times if the app is open and connected to the robot, then dorita980 cant connect and send commands. Close the apps and try again. Also try to soft reset your roomba (press 10 seconds the clean button)

200 and 304 are the response http status code. (200 ok, 304 not modified). the last number is the response length.

closed 2017-03-26 by koalazak

Mandy2000, 2017-03-26:
Hi,
I have definitely CLOSED the original app on iPhone AND iPad. It is definitely NOT running in the background. 
Funny that sometimes I can send 10 commands and ALL work and 2 hours later NO command is accepted ...

Also thanks for the explanation of the response