koalazak/dorita980 #43: Unable to get the password

NickWaterton, 2017-04-19:
I have read the similar issues listed here. but I don't understand how to fix this issue. I have followed the steps exactly (cloned the repo, cd to dorita980, npm install, npm run getpasswd 192.168.100.181).

I get the same output as issue #19 (which doe not give a solution by the way). Nodejs version is 0.10.25. Roomba version is 2.2.5-2. Output is:

`nick@proliant:~/Scripts/roomba/dorita980$ npm run getpasswd 192.168.100.181
npm ERR! Error: ENOENT, open '/home/nick/Scripts/roomba/dorita980/node_modules/getpasswd/package.json'
npm ERR! If you need help, you may report this log at:
npm ERR!     <http://github.com/isaacs/npm/issues>
npm ERR! or email it to:
npm ERR!     <npm-@googlegroups.com>

npm ERR! System Linux 4.4.0-72-generic
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "run" "getpasswd" "192.168.100.181"
npm ERR! cwd /home/nick/Scripts/roomba/dorita980
npm ERR! node -v v0.10.25
npm ERR! npm -v 1.3.10
npm ERR! path /home/nick/Scripts/roomba/dorita980/node_modules/getpasswd/package.json
npm ERR! code ENOENT
npm ERR! errno 34
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/nick/Scripts/roomba/dorita980/npm-debug.log
npm ERR! not ok code 0
`
What am I doing wrong? it must be something simple for those nodejs experts out there, but I'm a python/c person, so I don't know what I'm missing.

Thanks for any help...

NickWaterton, 2017-04-19:
OK, just upgraded nodejs to V6, and now it works. I think you should mention the versions of nodejs you need for these scripts to work...

closed 2017-04-19 by NickWaterton

koalazak, 2017-04-19:
the mention is here: https://github.com/koalazak/dorita980#note-for-nodejs-v010-users
RTFM :p