koalazak/dorita980 #45: Trying to understand how getpassword works

NickWaterton, 2017-05-05:
Hi, I've been using your excellent library, and have ported it to python. I'm now trying to figure out getpassword. Perhaps you could explain how it works (I'm just working on the fw V2 version).

I can get the blid, no problem.
This is the crucial part of the code:  
`
const packet = 'f005efcc3b2900';
  var client = tls.connect(8883, host, {rejectUnauthorized: false}, function () {
    client.write(new Buffer(packet, 'hex'));
  });

  client.on('data', function (data) {
    if (data.length === 2) {
      sliceFrom = 9;
      return;
    }
    if (data.length <= 7) {
      console.log('Error getting password. Follow the instructions and try again.');
    } else {
      console.log('Password=> ' + new Buffer(data).slice(sliceFrom).toString() + ' <= Yes, all this string.');
      console.log('Use this credentials in dorita980 lib :)');
    }
`

So what I would like to know is where  'f005efcc3b2900' comes from? I'm sending it as hex, and I get back two bytes 'f023' - what are you supposed to get back? is this an error code?

The other mystifying bit is:
`
if (data.length === 2) {
      sliceFrom = 9;
`
How can you slice from 9 if the data length is 2? Anyway, I'm assuming I should get more back than this, maybe I'm sending the "magic packet' incorrectly, and I'm getting an error code back.

Any tips or suggestions would be welcomed.

koalazak, 2017-05-07:
Hi, I dont know what `f005efcc3b2900` means exactly :p, But I know that the mobile application send this bytes to the robot in the process for getting up a password. The process is:
1) dorita send this packet to the bus
2) then the robot send the password to the bus.

That piece of code is a mess. There are some workarounds to be compatible with node 4, 5, 6 and 7 avoiding the diferences in each node versions handling buffers.

The thing with the length is: sometimes ( i dont remember in wich node version) the robot response 2 packets instead of one. In that case, the first one has 2 bytes of length and the second one contain the password but from byte 9. ( note the `return` sentence after `sliceFrom = 9;` that means this packet doesnt care and the next call to `client.on('data', cb)` has the password if length is more than 7).
But in other cases ( i dont remember in wich node version) the robot response with only one  packet with a length more than 7 , so here is the password slicing from 13 bytes. 
You miss this in your mentioned snippet:
```
function checkV2 () {
  var sliceFrom = 13;
```

So, i dont know how you are handling the connections with python, is a raw tls connection? or are you using a mqtt client? I thing all this `conversation` is out of the mqtt protocol.
You code is public? i can take a look.

ergards,

NickWaterton, 2017-05-07:
Thanks for the info, this explains a lot. So the magic packet is 7 bytes long, correct? I ask because the JavaScript description of Buffer(string, 'hex') is confusing.
I will try reading two packets if the first is just two bytes long (which is what I'm currently getting).

To get the password, I'm opening a raw TLS socket connection, but in the actual Python roomba class, I'm using Paho mqtt client with a TLS connection.

The Python class is working, I've just been putting the final touches to my live map drawing section.

The last part was how to get the password, so the Python library was fully featured.

I will post the code on my github, but I was waiting until I had all the bugs out before I posted. I'll let you know when it's up.

On a different note, any idea which ports roomba uses for firmware updates? I'm worried about an update closing off our mqtt access, but if I block all ports, the cloud service can't connect. I don't know if they are using the mqtt port for upgrades, but I would suspect not, more likely they are downloading over 433 (you could upgrade over mqtt though...). If I block 433, the cloud service is cut off, so that is used as part of the cloud service, but the same is true of 8883. Probably a combo of the two.

Anyway, thanks for the help.

rwscott1961, 2017-05-07:
The "magic" is just an MQTT Control Packet of type "Reserved", Remaining
Length is 5, and the contents 0xef, 0xcc, 0x3b, 0x29, 0x00.

Your response is an MQTT Control Packet of type "Reserved", 0x0f,
Remaining Length 0x23. The first 5 bytes of the response is the same
that was sent, 0xef, 0xcc, 0x3b, 0x29, 0x00, followed by 30 bytes of
password.

So the only "magic" is why does 0xef, 0xcc, 0x3b, 0x29, 0x00 mean "give
me your password". If your MQTT library allows you to specify the
Control Packet type, you should use that to send the magic. The response
would then, probably, be re-assembled correctly.

I attached my attempt at mapping.

NickWaterton, 2017-05-07:
Ok, now it makes sense.

I now have it working. I'll post the final thing in a day or so (still some bugs in my map drawing).

Thanks for all the help.