koalazak/dorita980 #46: Error with getpassword

ifongie, 2017-05-07:
Hi,

This week I've been having issue connecting to my Roomba through Homebridge. In the Home app, it says "No Response".

When I tried to run getpassword to get information from Roomba, I got the following error - I could get all information except the password. This was working before with the same Roomba with the same firmware. Any idea anybody???

Error log:
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node',
1 verbose cli   '/usr/local/bin/npm',
1 verbose cli   'run',
1 verbose cli   'getpassword',
1 verbose cli   '192.168.1.4' ]
2 info using npm@2.15.8
3 info using node@v4.4.7
4 verbose run-script [ 'pregetpassword', 'getpassword', 'postgetpassword' ]
5 info pregetpassword dorita980@3.0.11
6 info getpassword dorita980@3.0.11
7 verbose unsafe-perm in lifecycle true
8 info dorita980@3.0.11 Failed to exec getpassword script
9 verbose stack Error: dorita980@3.0.11 getpassword: `node ./bin/getpassword.js "192.168.1.4"`
9 verbose stack Exit status 1
9 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/lib/utils/lifecycle.js:217:16)
9 verbose stack     at emitTwo (events.js:87:13)
9 verbose stack     at EventEmitter.emit (events.js:172:7)
9 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/lib/utils/spawn.js:24:14)
9 verbose stack     at emitTwo (events.js:87:13)
9 verbose stack     at ChildProcess.emit (events.js:172:7)
9 verbose stack     at maybeClose (internal/child_process.js:827:16)
9 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
10 verbose pkgid dorita980@3.0.11
11 verbose cwd /Users/Fong/Desktop/dorita980
12 error Darwin 16.6.0
13 error argv "/usr/local/bin/node" "/usr/local/bin/npm" "run" "getpassword" "192.168.1.4"
14 error node v4.4.7
15 error npm  v2.15.8
16 error code ELIFECYCLE
17 error dorita980@3.0.11 getpassword: `node ./bin/getpassword.js "192.168.1.4"`
17 error Exit status 1
18 error Failed at the dorita980@3.0.11 getpassword script 'node ./bin/getpassword.js "192.168.1.4"'.
18 error This is most likely a problem with the dorita980 package,
18 error not with npm itself.
18 error Tell the author that this fails on your system:
18 error     node ./bin/getpassword.js "192.168.1.4"
18 error You can get information on how to open an issue for this project with:
18 error     npm bugs dorita980
18 error Or if that isn't available, you can get their info via:
18 error
18 error     npm owner ls dorita980
18 error There is likely additional logging output above.
19 verbose exit [ 1, true ]

koalazak, 2017-05-07:
Close all your mobile applications, update to latest node 4, and try again.

closed 2017-05-07 by koalazak

ifongie, 2017-05-08:
Updated to Node v6.10.3, still getting the same error. Any idea?

koalazak, 2017-05-08:
try removing node_modules folder and install again with `npm install`

ifongie, 2017-05-09:
Sorry mate, still not working. Do yo know if dorita980 works with firmware 2.2?

koalazak, 2017-05-09:
yes, works fine with <=2.2.5-2

ifongie, 2017-05-13:
Fixed - Had to do a factory reset on Roomba. Works now with Dorita980 and Homebridge!