koalazak/dorita980 #48: 690 & 890 compatible?

iosdeveloper, 2017-06-24:
Doesn't seem to work with the new 690. What about the 890?

koalazak, 2017-07-04:
Hi, I dont know. I dont own a 690/980. Anybody try?

iosdeveloper, 2017-07-26:
I checked with a 690 owner and it doesn't send the `pose` values.

pschmitt, 2017-07-26:
But the rest works?

iosdeveloper, 2017-07-26:
AFAIK, yes. No `pose` means no map though.

pschmitt, 2017-07-26:
That actually makes sense since only the 980 and the 960 have a camera and have the map feature on their spec sheet. The 890 behave similarly to the 690.

- 690: https://store.irobot.com/default/roomba-vacuuming-robot-vacuum-irobot-roomba-690/R690020.html
- 890: https://store.irobot.com/default/roomba-vacuuming-robot-vacuum-irobot-roomba-890/R890020.html
- 960: https://store.irobot.com/default/roomba-vacuuming-robot-vacuum-irobot-roomba-960/R960020.html
- 980: https://store.irobot.com/default/roomba-vacuuming-robot-vacuum-irobot-roomba-980/R980020.html

koalazak, 2017-07-26:
the only diference between 980 and 960 is the carpet bust and the battery.
here is the comparison chart:
http://www.irobot.com/For-the-Home/Vacuuming/Roomba.aspx#Comparison-Chart

pschmitt, 2017-07-26:
@iosdeveloper What does the `cap` field contain?

For my Roomba 980 that'd be the following:
```python
[('pose', 1),
  ('ota', 2),
  ('multiPass', 2),
  ('carpetBoost', 1),
  ('pp', 1),
  ('binFullDetect', 1),
  ('langOta', 1),
  ('maps', 1),
  ('edge', 1),
  ('eco', 1),
  ('svcConf', 1)]
```

maxbaeumle, 2017-07-26:
```"cap": {
    "eco": 1,
    "ota": 1,
    "svcConf": 1
}```

iosdeveloper, 2017-07-26:
```
"cap": {
    "eco": 1,
    "ota": 1,
    "svcConf": 1
}
```

pschmitt, 2017-07-27:
Oh so they are actually NOT set. Thanks for the info.

closed 2017-09-19 by koalazak