koalazak/dorita980 #50: Unable to get password on firmware 2.2.9-1

miceno, 2017-07-28:
Hi,

This library is really awesome! 

I would like to use it and I was using it several months ago, but now I can't get the password :(

Any suggestion?

Node version: v6.10.3
MacOSX Sierra 10.12.6
Dorita980 version:  HEAD@563be001e32dd479fd9caab0462ee77504c57fa4

```
dorita980$ npm run getpassword 192.168.1.33

> dorita980@3.0.11 getpassword dorita980
> node ./bin/getpassword.js "192.168.1.33"

Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Robot Data:
{ ver: '2',
  hostname: 'Roomba-3114491841613880',
  robotname: 'Moroco',
  ip: '192.168.1.33',
  mac: '74:C6:3B:C7:4D:27',
  sw: 'v2.2.9-1',
  sku: 'R980040',
  nc: 0,
  proto: 'mqtt',
  blid: '3114491841613880' }
Error getting password. Follow the instructions and try again.

```

pschmitt, 2017-07-28:
Not to promote my work but I litteraly just fixed a python implementation of the same function. You can give it a shot:

```
pip install git+https://github.com/pschmitt/roombapy@python3-fixes
roomba-getpassword
```

MrMxyzptlk, 2017-07-28:
pschmitt: I went to your github account, but couldn't find the repo?

pschmitt, 2017-07-28:
The link is just here: https://github.com/pschmitt/roombapy
By the way my PR just got merged, so please refer to the upstream repo: https://github.com/NickWaterton/Roomba980-Python

koalazak, 2017-07-28:
Hi @miceno,
I just try with your node version and works fine for me.
Same node and same firmware version.

```➜  dorita980 git:(master) nvm use 6.10.3
Now using node v6.10.3 (npm v3.10.10)
➜  dorita980 git:(master) node --version
v6.10.3
➜  dorita980 git:(master) npm run getpassword 192.168.1.104

> dorita980@3.0.11 getpassword /Users/facu/dev/dorita980
> node ./bin/getpassword.js "192.168.1.104"

Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Robot Data:
{ ver: '2',
  hostname: 'Roomba-xxxxxxxxxxxxxxxx',
  robotname: 'Dorita',
  ip: '192.168.1.104',
  mac: 'xx:xx:xx:xx:xx:xx',
  sw: 'v2.2.9-1',
  sku: 'R98----',
  nc: 0,
  proto: 'mqtt',
  blid: 'xxxxxxxxxxxxxxx' }
Password=> xxxxxxxxxxxxx <= Yes, all this string.
Use this credentials in dorita980 lib :)```

try deleting node_modules and reinstall. Or updating node.js version. Or try with node 4.

let me kwno if it works.

closed 2017-08-12 by koalazak