koalazak/dorita980 #51: Issue Dorita980 getpassword

ecandamo, 2017-09-02:
Hi,
First of all thanks for this repo.
I am having issues getting my Roomba's information.
I get the following error as per the log
-------------------------------------------------------------
0 info it worked if it ends with ok
1 verbose cli [ 'C:\\Program Files\\nodejs\\node.exe',
1 verbose cli   'C:\\Users\\esteb\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js',
1 verbose cli   'run',
1 verbose cli   'getpassword',
1 verbose cli   '10.0.0.234' ]
2 info using npm@5.4.0
3 info using node@v6.11.2
4 verbose run-script [ 'pregetpassword', 'getpassword', 'postgetpassword' ]
5 info lifecycle dorita980@3.0.11~pregetpassword: dorita980@3.0.11
6 info lifecycle dorita980@3.0.11~getpassword: dorita980@3.0.11
7 verbose lifecycle dorita980@3.0.11~getpassword: unsafe-perm in lifecycle true
8 verbose lifecycle dorita980@3.0.11~getpassword: PATH: C:\Users\esteb\AppData\Roaming\npm\node_modules\npm\bin\node-gyp-bin;C:\Users\esteb\dorita980\node_modules\.bin;C:\Users\esteb\bin;C:\Program Files\Git\mingw64\bin;C:\Program Files\Git\usr\local\bin;C:\Program Files\Git\usr\bin;C:\Program Files\Git\usr\bin;C:\Program Files\Git\mingw64\bin;C:\Program Files\Git\usr\bin;C:\Users\esteb\bin;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\Program Files (x86)\Skype\Phone;C:\Users\esteb\AppData\Local\Microsoft\WindowsApps;C:\adb;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\Program Files\PuTTY;C:\Program Files\Git\cmd;C:\Program Files\nodejs;C:\Users\esteb\AppData\Local\Microsoft\WindowsApps;C:\Users\esteb\AppData\Roaming\npm;C:\Program Files\Git\usr\bin\vendor_perl;C:\Program Files\Git\usr\bin\core_perl
9 verbose lifecycle dorita980@3.0.11~getpassword: CWD: C:\Users\esteb\dorita980
10 silly lifecycle dorita980@3.0.11~getpassword: Args: [ '/d /s /c', 'node ./bin/getpassword.js "10.0.0.234"' ]
11 silly lifecycle dorita980@3.0.11~getpassword: Returned: code: 1  signal: null
12 info lifecycle dorita980@3.0.11~getpassword: Failed to exec getpassword script
13 verbose stack Error: dorita980@3.0.11 getpassword: `node ./bin/getpassword.js "10.0.0.234"`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (C:\Users\esteb\AppData\Roaming\npm\node_modules\npm\node_modules\npm-lifecycle\index.js:280:16)
13 verbose stack     at emitTwo (events.js:106:13)
13 verbose stack     at EventEmitter.emit (events.js:191:7)
13 verbose stack     at ChildProcess.<anonymous> (C:\Users\esteb\AppData\Roaming\npm\node_modules\npm\node_modules\npm-lifecycle\lib\spawn.js:55:14)
13 verbose stack     at emitTwo (events.js:106:13)
13 verbose stack     at ChildProcess.emit (events.js:191:7)
13 verbose stack     at maybeClose (internal/child_process.js:891:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
14 verbose pkgid dorita980@3.0.11
15 verbose cwd C:\Users\esteb\dorita980
16 verbose Windows_NT 10.0.15063
17 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\esteb\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "run" "getpassword" "10.0.0.234"
18 verbose node v6.11.2
19 verbose npm  v5.4.0
20 error code ELIFECYCLE
21 error errno 1
22 error dorita980@3.0.11 getpassword: `node ./bin/getpassword.js "10.0.0.234"`
22 error Exit status 1
23 error Failed at the dorita980@3.0.11 getpassword script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
------------------------
Kindly let me know what I a doing wrong and how do I solve it. I am following step by step the readme.

JohanSF, 2017-09-02:
I have the same problem. In the terminal the first error is:
npmERR! Error: ENOENT, open 

'/mnt/c/Users/Johan/Desktop/rest980/dorita980/node_modules/getpassword/package.json'

Could the problem be that there is no 'getpassword' folder in the 'node_modules' folder? and how do I solve it?

koalazak, 2017-09-02:
Are you following all this steps?
```
$ git clone https://github.com/koalazak/dorita980.git
$ cd dorita980
$ npm install
$ npm run getpassword <robotIP>
```

closed 2017-09-11 by koalazak