koalazak/dorita980 #55: Get password issue...

MrMxyzptlk, 2017-10-11:
I have successfully used you excellent utility on one of my Linux boxes several times in the last year.
I have updated nodejs to the latest, and have the latest version of your utility.

When I run the the get password command this is the output...with the BLID obfuscated.
Not sure why it won't work this time...any idea?

Robot Data:
{ ver: '2',
  hostname: 'Roomba-30F7012451143700',
  robotname: 'Celeste',
  ip: '10.0.0.154',
  mac: '28:C2:DD:8A:8C:EE',
  sw: 'v2.2.11-4',
  sku: 'R98----',
  nc: 0,
  proto: 'mqtt',
  blid: 'xxxxxxxxxxxxxxxxxxx' }
Error getting password. Follow the instructions and try again.

koalazak, 2017-10-16:
Hi, what node version do you have?
close your mobile app

MrMxyzptlk, 2017-10-16:
nodejs version 4.2.6

koalazak, 2017-10-16:
that is not the latest on 4's branch. Try updating to Node.js 4.8.4. Or latest 5 or latest 6.
And take a look at this closed issues:
#34
#27 (this in particular)
#21
#19

closed 2017-10-16 by koalazak

MrMxyzptlk, 2017-10-16:
Thank you, I had updated to the latest before I posted this error, but I guess "sudo apt-get update" and "upgrade", didn't do the trick. I will try again...

MrMxyzptlk, 2017-10-16:
That did the trick...somehow my password changed from one of the last Roomba firmware updates.
I'm now on 8.xxxx node.js

Thank you for your time!