koalazak/dorita980 #56: Brand new 690 Firmware ( 3.2.7+26 )

Norien, 2017-10-15:
Just picked up a brand new 690 and I’m having a hell of a time getting my Id and password.
Is there any chance for support for newer firmware?

koalazak, 2017-10-16:
Hi,
If my 980 eventualy updated the firmware, I will update dorita980 to support the new firmware.
But at the moment my 980 has v2.2.11-4

koalazak, 2017-10-20:
there is not firmware v3 for 900 series.
https://homesupport.irobot.com/app/answers/detail/a_id/529/p/4116

closed 2017-10-20 by koalazak

Norien, 2017-10-20:
This is from my iRobot app, I assumed that was the firmware version. If not where can I find the correct version?

[image]

koalazak, 2017-10-20:
yes, 690 has a completly diferent firmware (v3.x.x) that 900 series (2.x.x).
So, dorita980 is not going to upgrade to be compatible with other firmwares that are not of the 800/900 series.

Norien, 2017-10-20:
I'm considering exchanging my 960 for a 980 so this may just push me to do so. Thank you for your time.