koalazak/dorita980 #58: New app adds ifttt - is internet access still safe?

sheppy99, 2017-11-20:
The iOS App Store has an iRobot update that allows ifttt to trigger a roomba. Is it safe to open up the robot to internet access and any likely firmware update?

ristomatti, 2017-11-28:
@sheppy99 Note that the support is only for US and England at least for now. I believe the FAQ said the support is global after it was announced in the app description but either they have now updated the info or I remember wrong since it now doesn't. At least for me the newest firmware does not work with dorita980 so if it works for you double check the availability if control through this module is important for you.

I have the model 966 though which might be the cause also. I just believe the 900 series models all have the same firmware. When I first started the robot it was running 1.x.x firmware and I was at least able to fetch the password using dorita980 but after it got updated it doesn't seem to connect anymore.

Maybe I should setup my router to direct the Robot to use a VPN connection to UK to at least get the IFTTT features working since I live in Finland. 😏

koalazak, 2017-12-10:
firmware 3.2.10 is not supported yet. Avoid the firmware update. 
If they allow the firmware update outside US/UK, I will work on make dorita980 compatible again.

closed 2017-12-10 by koalazak

sheppy99, 2017-12-11:
In New Zealand I can add the iRobot channel to IFTTT, but my robot isn’t listed as a device so it’s useless. Fortunately dorita980 still works although if I leave it allowed to access the internet it seems to stop responding after a while until I block it and reset the robot