koalazak/dorita980 #59: New Firmware locked out again....

Harald-B-Rupp, 2017-12-01:
Hi All,

will Firmware Version 3.2.10 for the 600 and 800 Series also supported?

Thanks

Harald 

{ ver: '3',
  hostname: 'Roomba-3162880891126810',
  robotname: 'HE89501',
  ip: '192.168.7.213',
  mac: '40:9F:38:93:25:89',
  sw: '3.2.10',
  sku: 'R895040',
  nc: 1,
  proto: 'mqtt',
  cap: { ota: 1, eco: 1, svcConf: 1 } }

koalazak, 2017-12-10:
Firmware 3.2.10 is not supported. (I dont know what works and what doest work on that version, because the upgrade is only for UK or new 8xx series)

closed 2017-12-10 by koalazak

Harald-B-Rupp, 2017-12-11:
Hi and Thanks for your Answer.

I got it working with Start. Stop, Pause, Docking, Resume!

Mission, Batterie Status and all other Info Stuff is not anymore working.

Harald