koalazak/dorita980 #60: Can't find /node_module/getpassword/package.json when trying to get blid and

airjesse123, 2017-12-06:
Alright I'm feeling like a bit of an idiot. I am having the same issue as Melkor82 had in Issue12, even though I've cloned the repo.
I am in the directory `/home/pi/Documents/dorita980` and have already run `npm install`.
When I run `npm run getpassword 192.168.1.22`

I get
```

npm ERR! Error: ENOENT: no such file or directory, open '/home/pi/Documents/dorita980/node_modules/getpassword/package.json'
npm ERR!     at Error (native)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.9.59-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "run" "getpassword" "192.168.1.22"
npm ERR! cwd /home/pi/Documents/dorita980
npm ERR! node -v v4.8.2
npm ERR! npm -v 1.4.21
npm ERR! path /home/pi/Documents/dorita980/node_modules/getpassword/package.json
npm ERR! syscall open
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/Documents/dorita980/npm-debug.log
npm ERR! not ok code 0
```

koalazak, 2017-12-10:
update your npm version to v3 or more.
Also you can upgrade your node version to the latest 6 (that include latest npm3)

closed 2017-12-10 by koalazak

bpgregson, 2018-12-26:
I am experiencing the same issue, returning the following error:

> 
> pi@myPi:~/dorita980 $ npm run getpassword 192.168.0.37
> (node:6613) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
> npm ERR! Error: ENOENT: no such file or directory, open '/home/pi/dorita980/node_modules/getpassword/package.json'
> npm ERR! If you need help, you may report this *entire* log,
> npm ERR! including the npm and node versions, at:
> npm ERR!     <http://github.com/npm/npm/issues>
> npm ERR! System Linux 4.14.52-v7+
> npm ERR! command "/usr/bin/node" "/usr/bin/npm" "run" "getpassword" "192.168.0.37"
> npm ERR! cwd /home/pi/dorita980
> npm ERR! node -v v8.11.1
> npm ERR! npm -v 1.4.21
> npm ERR! path /home/pi/dorita980/node_modules/getpassword/package.json
> npm ERR! syscall open
> npm ERR! code ENOENT
> npm ERR! errno -2
> npm ERR!
> npm ERR! Additional logging details can be found in:
> npm ERR!     /home/pi/dorita980/npm-debug.log
> npm ERR! not ok code 0

Any suggestions on how to resolve?  Thanks.

koalazak, 2018-12-26:
> update your npm version to v3 or more.

:point_up_2:

bpgregson, 2018-12-27:
> > update your npm version to v3 or more.
That was the problem, issue resolved, thank you.