koalazak/dorita980 #61: Connected but without response

wojtulab, 2017-12-06:
hello,
I have problem with my Roomba980.
After I get blid & password, I've tried many times to control it via node scripts without success.

My roomba details:
```
Robot Data:

{ ver: '2',
  hostname: 'Roomba-6945CXXXXXXX840',
  robotname: 'Roomba',
  ip: '192.168.1.3',
  mac: '80:A1:89:11:1D:FD',
  sw: 'v2.2.11-4',
  sku: 'R98----',
  nc: 0,
  proto: 'mqtt',
  blid: '6945CXXXXXXX840' }
Password=> AsGzX0pzA1EReTYh <= Yes, all this string.
Use this credentials in dorita980 lib :)
```

koalazak, 2017-12-10:
Can you share the steps and code you are running?

wojtulab, 2017-12-12:
Sure, the first what i tested is test.js:
root@orangepizero:~/apps/dorita980# node test.js
I have to wait ... 30min later i cancel it becase there wasn't any response.
```

var dorita980 = require('dorita980');

var myRobotViaLocal = new dorita980.Local('6945CXXXXXXX840', 'AsGzX0pzA1EReTYh', '192.168.1.3'); // robot IP address

myRobotViaLocal.on('connect', init);

function init () {
  myRobotViaLocal.getWeek()
  .then((weekConfig) => {
    console.log(weekConfig)
    myRobotViaLocal.end()
  })
  .catch(console.log);
}

```

koalazak, 2017-12-12:
yes, try `nc -vz 192.168.1.3 8883`

wojtulab, 2017-12-12:
port seems to be closed?

`root@orangepizero:~/cron# nc -vz 192.168.1.3 8883
nnetfd reuseport failed : Protocol not available
roomba.lan [192.168.1.3] 8883 (?) : Connection refused
`

koalazak, 2017-12-12:
yes, you can not reach the required port on roomba. 
Maybe a firewall in your router?
Or if you have the HOME mobile app opened, please close it and try again.

wojtulab, 2017-12-12:
root@orangepizero:~/cron# nmap 192.168.1.3

Starting Nmap 6.47 ( http://nmap.org ) at 2017-12-12 13:54 CET
Nmap scan report for roomba.lan (192.168.1.3)
Host is up (0.0012s latency).
All 1000 scanned ports on roomba.lan (192.168.1.3) are closed
MAC Address: 80:A5:89:71:9D:FD (Unknown)

Internal firewall is closed.
All apps are closed also.

koalazak, 2017-12-12:
sounds like a network issue, you should reach the 8883 port on the robot. Digg on that connectivity issue and then try again with dorita980

closed 2017-12-12 by koalazak

wojtulab, 2017-12-12:
Why the network issue?
I am trying to connect directly from my router (no firewalls etc) and port is also closed?

koalazak, 2017-12-12:
try restarting your robot (press home buton for 10s).
This is how looks a `nc` to the robot without a connectivity issue:
```
nc -vz 192.168.1.104 8883
found 0 associations
found 1 connections:
     1:	flags=82<CONNECTED,PREFERRED>
	outif en0
	src 192.168.1.101 port 52883
	dst 192.168.1.104 port 8883
	rank info not available
	TCP aux info available

Connection to 192.168.1.104 port 8883 [tcp/*] succeeded!
```

wojtulab, 2017-12-12:
You are right but after restart procedure nothing changed in my problem.

koalazak, 2017-12-12:
sorry, press the clean buton for 10s, not the home button

wojtulab, 2017-12-12:
ok, restarted.

```
root@router:~# nc 192.168.1.3 8883
nc: can't connect to remote host (192.168.1.3): Connection refused
```

wojtulab, 2017-12-12:
I disabled firewall on my router. The status is the same...connection refused on 8883 tcp port.

wojtulab, 2017-12-13:
finally I resolved the issue.

Problem wasn't connected with my WiFi / Router etc.
I remove robot from my mobile app and reconfigure it again. After that I managed to run scripts.

Greetings