koalazak/dorita980 #62: Error Endpoint not found

Mandy2000, 2018-02-06:
Hi,
after it worked fine for many weeks now I very often can't start Roomba using the URL http://192.168.1.1:3000/api/local/action/start. The robot just isn't responding.
I tried to reinstall the repository but that did not solve my problem

Now I checked the terminal and I realized that there is an error with the endpoint. When I start REST980 I get the error shown below a few seconds after starting.

Funny that sometimes it works (but with a very long delay) ... see last line "start 304 2186.455 ms"

what can I do please?
thanks for help
Mandy

iMac:rest980 mymachine$ DEBUG=rest980:* npm start

> rest980@2.0.6 start /Users/mymachine/rest980
> node ./bin/www

  rest980:server Listening on port 3000 +0ms
GET /favicon.ico 404 3.962 ms - 2446
Error: Endpoint not found.
    at /Users/mymachine/rest980/app.js:62:13
    at Layer.handle [as handle_request] (/Users/mymachine/rest980/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/Users/mymachine/rest980/node_modules/express/lib/router/index.js:317:13)
    at /Users/mymachine/rest980/node_modules/express/lib/router/index.js:284:7
    at Function.process_params (/Users/mymachine/rest980/node_modules/express/lib/router/index.js:335:12)
    at next (/Users/mymachine/rest980/node_modules/express/lib/router/index.js:275:10)
    at /Users/mymachine/rest980/node_modules/express/lib/router/index.js:635:15
    at next (/Users/mymachine/rest980/node_modules/express/lib/router/index.js:260:14)
    at Function.handle (/Users/mymachine/rest980/node_modules/express/lib/router/index.js:174:3)
    at router (/Users/mymachine/rest980/node_modules/express/lib/router/index.js:47:12)
    at Layer.handle [as handle_request] (/Users/mymachine/rest980/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/Users/mymachine/rest980/node_modules/express/lib/router/index.js:317:13)
    at /Users/mymachine/rest980/node_modules/express/lib/router/index.js:284:7
    at Function.process_params (/Users/mymachine/rest980/node_modules/express/lib/router/index.js:335:12)
    at next (/Users/mymachine/rest980/node_modules/express/lib/router/index.js:275:10)
    at authHandler (/Users/mymachine/rest980/app.js:23:48)
    at Layer.handle [as handle_request] (/Users/mymachine/rest980/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/Users/mymachine/rest980/node_modules/express/lib/router/index.js:317:13)
    at /Users/mymachine/rest980/node_modules/express/lib/router/index.js:284:7
    at Function.process_params (/Users/mymachine/rest980/node_modules/express/lib/router/index.js:335:12)
    at next (/Users/mymachine/rest980/node_modules/express/lib/router/index.js:275:10)
    at SendStream.error (/Users/mymachine/rest980/node_modules/serve-static/index.js:121:7)
GET /api/local/action/start 304 2186.455 ms - -

koalazak, 2018-02-14:
the `Error: Endpoint not found.` error is for the `GET /favicon.ico`. the browser is requiring the favicon when you visit the page but there is no favicon so you get 404 for that request. Forget that error, it is not related with the problem.

So, about the real problem (you get a timeout calling the `/api/local/action/start` endpoint) seems like your server cant stablish a connection with your robot.
What robot model you have? and what firmware version you have? and what node.js version?

Do you tried the simple example of dorita980? do you have the robot password and ip ok?

Mandy2000, 2018-02-14:
Hi,

thanks for contacting me. 3 days after I sent you the mail the connection with the robot suddenly started working again. Maybe there was a problem on the iRobot side ...

greetings
Manfred

closed 2018-02-14 by koalazak