koalazak/dorita980 #63: Roomba 695

joydashy, 2018-02-14:
Is this supposed to be compatible with Roomba 695? Read some different opinions on the matter. If not, then I guess I bought the wrong Roomba :(

If it is supposed to work. Am running into this problem:

```
Robot Data:
undefined
```

Debug log shows:

```
4 verbose run-script [ 'pregetpassword', 'getpassword', 'postgetpassword' ]
5 info lifecycle dorita980@3.0.12~pregetpassword: dorita980@3.0.12
6 info lifecycle dorita980@3.0.12~getpassword: dorita980@3.0.12
7 verbose lifecycle dorita980@3.0.12~getpassword: unsafe-perm in lifecycle true
8 verbose lifecycle dorita980@3.0.12~getpassword: PATH: /usr/lib/node_modules/npm/bin/node-gyp-bin:/home/redbull/roomba/dorita980/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl
9 verbose lifecycle dorita980@3.0.12~getpassword: CWD: /home/redbull/roomba/dorita980
10 silly lifecycle dorita980@3.0.12~getpassword: Args: [ '-c', 'node ./bin/getpassword.js "192.168.10.166"' ]
11 silly lifecycle dorita980@3.0.12~getpassword: Returned: code: 1  signal: null
12 info lifecycle dorita980@3.0.12~getpassword: Failed to exec getpassword script
13 verbose stack Error: dorita980@3.0.12 getpassword: `node ./bin/getpassword.js "192.168.10.166"`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:280:16)
13 verbose stack     at emitTwo (events.js:125:13)
13 verbose stack     at EventEmitter.emit (events.js:213:7)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at emitTwo (events.js:125:13)
13 verbose stack     at ChildProcess.emit (events.js:213:7)
13 verbose stack     at maybeClose (internal/child_process.js:897:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:208:5)
14 verbose pkgid dorita980@3.0.12
15 verbose cwd /home/redbull/roomba/dorita980
16 verbose Linux 4.13.4-1-pve
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "run" "getpassword" "192.168.10.166"
18 verbose node v8.1.2
19 verbose npm  v5.5.1
20 error code ELIFECYCLE
21 error errno 1
22 error dorita980@3.0.12 getpassword: `node ./bin/getpassword.js "192.168.10.166"`
22 error Exit status 1
23 error Failed at the dorita980@3.0.12 getpassword script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

koalazak, 2018-02-14:
what firmware version do you have on your 695 robot?

joydashy, 2018-02-14:
Running the latest I believe: 3.2.10+32

joydashy, 2018-02-16:
Solved this by switching to IFTTT instead!

closed 2018-02-16 by joydashy