koalazak/dorita980 #65: Lightweight alternative?

eduperez, 2018-03-27:
I am planning to control a Roomba from an embedded device. So far, dorita980 is working perfectly (thanks for all your work!), but I think that node.js is a bit too heavy for my setup, and would like to search for something lighter.

I mostly need to be able to turn the Roomba on and off, nothing specially fancy, so I just need to send commands to the Roomba. I have been experimenting with `mosquitto`'s MQTT client, but so far I could not even connect to the Roomba (there is something wrong with the TLS, I guess).

Does anybody know of any alternatives to dorita980, that do not require node.js? Or how to send direct MQTT commands to the Roomba? Many thanks!

koalazak, 2018-03-27:
did you evaluate install [rest980](https://github.com/koalazak/rest980) on your lan, and then just sent http requests from your lightweight setup to rest980?

eduperez, 2018-03-27:
Yes, I had a look at rest980; but it also needs node.js, and I am trying to find a solution that does not involve having any "heavy" device running 24/7. Thanks for the pointer.

koalazak, 2018-03-27:
ok. And did you disable the certificate validation using mosquitto?
https://mosquitto.org/api/files/mosquitto-h.html#mosquitto_tls_opts_set
also check this methods:

mosquitto_tls_set | Configure the client for certificate based SSL/TLS support.
```
mosquitto_tls_insecure_set | Configure verification of the server hostname in the server certificate.
mosquitto_tls_opts_set | Set advanced SSL/TLS options.
```

You need to setup mosquito to ignore the error validating the TLS roobot's certificate on connect.

ristomatti, 2018-03-27:
Sorry to also take part of this off topic conversation but could I somehow send commands using MQTT? I never got dorita980 working unfortunately. I would also be happy with just start/stop/return home.

The official IFTTT support would also be ok (although a bit more cumbersome to integrate to Node-RED) for me if they'd just expand it outside US/UK. I've contacted iRobot support but they had no timeline to give. :unamused:

eduperez, 2018-03-28:
I am using mosquitto's utilities (mosquitto_pub and mosquitto_sub), not the mosquitto library, and I could not guess how to pass those parameters from the utils; Is should probably bit the bullet and build my own client... however, I noticed that the global "node_modules" folder was the only roadblock in my setup, everything else could be accommodated in my embedded device; so I moved that folder to an external drive, and now I can run node.js and dorita980 successfully.

Many thanks for your help.

closed 2018-03-28 by eduperez