koalazak/dorita980 #71: Support for 600 Wifi enabled series / other Roombas

ahorseman, 2018-09-13:
Hello,

I wanted to ask whether you guys are also planning to support other Roombas than just the 980?

The Wifi enabled 600 series (currently on a firmware 3.x) would also be interesting to support, as they are much more for the mass market (given the price) and I would be happy to help discovering supported features and requests.

Is there a way to get an API overview of the robot's API (something like wsdl was for web services) that I can post here for enhancing (or forking) dorita980 for 600 series?

koalazak, 2018-12-03:
the best way is reversing the mobile app and sniffing the netword trafic between the mobile app and roomba or cloud. I dont own a 600 robot to try.
Let me know if you start a 600's fork!
thanks!

Zefau, 2018-12-03:
I don't own a Roomba 600 or any other with v3 firmware, but according to information in the ioBroker forum, the v3 software uses MQTT via port 8883.
See https://forum.iobroker.net/viewtopic.php?f=20&t=8092&p=199686#p199367

Zefau, 2018-12-03:
@koalazak How are you sniffing the network traffic? What software are you using and how do you reverse?

koalazak, 2018-12-04:
im sniffing with wireshark and charles proxy in mac. And for reversing the app I grab the .apk file from Android store and decompile it with online tools.

crashf, 2018-12-18:
Im not sure if this will help you or not, but, I spent some time sniffing packets with wireshark. Here is a dump of just traffic from/to the roomba 675. Opened the app and performed a "return to duck" request.

[filtereddump-roomba675-fromrouter.zip](https://github.com/koalazak/dorita980/files/2690562/filtereddump-roomba675-fromrouter.zip)

crashf, 2018-12-18:
I will also try to dump during the wireless setup etc later today. I am not great with python so if anyone is up for picking around let me know what else I can dump.

koalazak, 2019-01-04:
Hi, sorry for the delay. That dumps are ssl encryted. Did you sign the trafic with your own CA? you need to do that to decrypt the dump content

Jeff-Cortese, 2019-01-06:
Hi, I was able to get this library working with my Roomba 690 (firmware 3.2). A tip that might be useful is that you can get the Roomba's IP address from the iRobot HOME app by tapping `More (...) > Settings > Wi-Fi Settings > Robot Wi-Fi Details`. Using that IP, I could use the `getpassword` npm script to look up the blid and password (via `checkV2()` function in `bin/getpassword.js`).

With that...
```JavaScript
const dorita980 = require('dorita980');
const myLocalRobot = new dorita980.Local('blid_here', 'password_here', 'ip_here');
myLocalRobot.on('connect', () => { console.log('connected'); });
```
...works as expected

Zefau, 2019-01-06:
The connection works with Firmware v3.0, but no data is received as far as I know.
Do you have a running instance of ioBroker? I made an adapter for Roomba on this plattform (see https://github.com/Zefau/ioBroker.roomba). 

Other users report, that a connection is successful with Firmware v3, but no data is received.

Are you able to execute any commands after connection or able to retrieve preferences?

Kind regards,
Zefau

Jeff-Cortese, 2019-01-07:
@Zefau I'm not sure what ioBroker is, so probably not if it's not a part of dorita and running as a side-effect of the code I posted earlier. And yes, a few commands did work (I have only tried `start`, `pause`, `resume`, and `end`).

@koalazak I tweaked it a little to format the packet payload buffer:
```JavaScript
var dorita980 = require('dorita980');

(async () => {
  const myLocalRobot = new dorita980.Local('blid_here', 'password_here', 'ip_here');

  myLocalRobot.on('connect', () => { 
    console.log('connected'); 
    myLocalRobot.on('packetreceive', packet =>  {
      packet.payload = JSON.parse(packet.payload);
      console.log(JSON.stringify(packet, null, 2));
    });
  });
})()
```
The output is as follows (I redacted some info using X's because I'm not sure what is sensitive info).
Packets started to repeat, so I terminated it after about 10 seconds. I can run it longer if you'd like.
```
connected
{
  "cmd": "publish",
  "retain": false,
  "qos": 0,
  "dup": false,
  "length": 182,
  "topic": "wifistat",
  "payload": {
    "state": {
      "reported": {
        "netinfo": {
          "dhcp": true,
          "addr": xxxxxxxxxx,
          "mask": xxxxxxxxxx,
          "gw": xxxxxxxxxx,
          "dns1": xxxxxxxxxx,
          "dns2": xxxxxxxxxx,
          "bssid": "xxxxxxxxxx",
          "sec": 4
        }
      }
    }
  }
}
{
  "cmd": "publish",
  "retain": false,
  "qos": 0,
  "dup": false,
  "length": 78,
  "topic": "wifistat",
  "payload": {
    "state": {
      "reported": {
        "wifistat": {
          "wifi": 1,
          "uap": false,
          "cloud": 1
        }
      }
    }
  }
}
{
  "cmd": "publish",
  "retain": false,
  "qos": 0,
  "dup": false,
[... 346 lines cut ...]
  "dup": false,
  "length": 65,
  "topic": "wifistat",
  "payload": {
    "state": {
      "reported": {
        "signal": {
          "rssi": -34,
          "snr": 56
        }
      }
    }
  }
}
{
  "cmd": "pingresp",
  "retain": false,
  "qos": 0,
  "dup": false,
  "length": 0,
  "topic": null,
  "payload": null
}

```

koalazak, 2019-01-07:
Hi, @Jeff-Cortese, thanks a lot! this was very helpful.
I think the problem with the other features is that dorita980 is waiting for the `pose` state and your robot (600 series) is not reporting that postion, then dorita980 wait forever.
I will work on fix this, seems easy to do. If you want to test the fix just open `node_modules/dorita980/lib/v2/local.js` and remove the occurences of `pose` string or objects (like line 35, 36, 107, 109, etc)

thanks again guys,

gik007, 2019-01-07:
Hi @koalazak,

> im sniffing with wireshark and charles proxy in mac. And for reversing the app I grab the .apk file from Android store and decompile it with online tools.

I know that your project is focused on the 980 model (or it was like that at the beginning). I have another model (e5) and have tried to use your library without success. I have the impression that your implementation is close to working with that model, but I fail to get the password ("Error getting password. Follow the instructions and try again." which hints at a short response to the MQTT request. The response I get is "f005efcc3b2903", which is pretty close to the request your code sends to get the password. 

I could try to reverse engineer the network traffic, but where should I start? Your implementation shows that you have understood pretty well what happens between client and server (at least for your model), do you have that information somewhere so that we can start creating an unofficial documentation of the communication used by the different roomba models?

Gracias!

Zefau, 2019-01-07:
We discussed e5 support in https://github.com/Zefau/ioBroker.roomba/issues/1. Remove "pose" from preferences worked here as well.

gik007, 2019-01-07:
Good news. The error was sitting between keyboard and chair. I did not correctly understand the message about pressing the HOME button (I was pressing the "CLEAN" button). There was a message in another thread giving that hint and it was helpful. So I can confirm that things are working well on the e5. Thanks for making it possible!

koalazak, 2019-01-08:
600 and e5 series supported in dorita980@3.1.0
thanks guys

closed 2019-01-08 by koalazak

TonyBrobston, 2019-07-19:
I can confirm that I was able to get the blid and password on a Roomba 690 version 3.2.40. 

It's super important that the wifi symbol flashes. I had to hold the button above the "Clean" button and the button below the "Clean" button for 2 seconds to get the wifi symbol to flash. Any run before that generated all the information except the password.

It took me forever to figure out the wifi symbol not blinking was the issue, I figured I needed to share this somewhere, hope it helps someone else!

nini57, 2019-10-19:
hello
can me confirm this is right please ?

LINE 35    ` if (robotState.cleanMissionStatus) { //) { && robotState.pose) {`
LINE 36   ` client.emit('mission', filterProps(['cleanMissionStatus', 'bin'])); 'bin']));`
LINE 107 `    getPreferences: (decode) => waitPreferences(decode, ['cleanMissionStatus', 'cleanSchedule', 'name', 'vacHigh', 'signal'], false),
`
LINE 109 `    getMission: (decode) => waitPreferences(decode, ['cleanMissionStatus', 'bin', 'batPct'], true),
`