koalazak/dorita980 #72: Actions are not working with firmware 2.4.6-3

tobiasramin, 2018-09-22:
Hi,
first of all, thanks for your effort to create this wundeful library. It cost me only a hour to get the Rest-Server running.
I can get preferences, mission etc, but all mentioned Actions are not working.

I used myapp.js, but did not work and I dont get an error message or something:

```

var dorita980 = require('dorita980');

var myRobotViaLocal = new dorita980.Local('6944401882027730', 'Bb1gOunJ3dxoOcp0', '192.168.1.4'); // robot IP address 

myRobotViaLocal.on('connect', init);

function init () {
  myRobotViaLocal.start()
  .then(() => myRobotViaLocal.end()) // disconnect to leave free the channel for the mobile app.
  .catch(console.log);
}
```

iRobot Firmware: 2.4.6-3

I double checked with the provided Rest980 Server, but even there no success. Is that an issue with the Firmware?

Do you have an idea?

tobiasramin, 2018-09-23:
Hi @koalazak,
thanks for your Response. I just startet the Rest980 container on my Synology again and sent an Get http://syn:3000/api/local/action/start and the iRobot started. Afterwards I stopped him with the stop Action. That was successful. Unfortunately resume or start again is not working. After I restarted the Container again, start is working again. Unfortunately no log entries mentioning anything, only the Feedback About the executed Action.

Here my robot Details:
`Robot Data:
{ ver: '3',
  hostname: 'Roomba-6944401882027730',
  robotname: 'Boomba',
  ip: '192.168.1.4',
  mac: '28:C2:DD:8A:CE:F5',
  sw: 'v2.4.6-3',
  sku: 'R98----',
  nc: 0,
  proto: 'mqtt',
  cap:
   { pose: 1,
     ota: 2,
     multiPass: 2,
     carpetBoost: 1,
     pp: 1,
     binFullDetect: 1,
     langOta: 1,
     maps: 1,
     edge: 1,
     eco: 1,
     svcConf: 1 },
  blid: '6944401882027730' }`

For the Rest980, I used your docker Image and start it with the folowing variales:

[image]

Alternatively I installed node.js v8.12.0 on Windows 10. Same behaviour.

The sporadical run makes me worry...

koalazak, 2018-09-24:
put `FIRMWARE_VERSION` in `2` and `ENABLE_LOCAL` in `yes`

also, remove your robot user/password from here. Everybody may start playing with your robot via remote control using your credentials...