koalazak/dorita980 #75: Can't catch thrown error

Zefau, 2018-11-18:
Hi,

thanks for you library. I'm using it for an ioBroker module implementation (see https://github.com/Zefau/ioBroker.roomba).

In the course of my implementation I'm struggling to catch the error thrown in lib/v2/local.js at row 30: https://github.com/koalazak/dorita980/blob/df31a27e8f175836ba092620eff61e7ff5ea340d/lib/v2/local.js#L30
Try-catch-block fails catching the error, e.g. following code will NOT work:

`	try
	{
		robot = new Roomba.Local(adapter.config.username, library.decode(settings.decode.key, adapter.config.password), adapter.config.ip); // username, password, ip
	}
	catch(err)
	{
		adapter.log.debug(err.message);
		return;
	}`

My suggestion is to not throw the error and leave error-handling it to the end-point-implementation.

Thanks
Zefau

koalazak, 2018-12-03:
I see the problem. But Im afraid of change that and break currect implementations. Maybe we can merge this change with a major version update. But that require some addicional changes in readme and package.json.
PRs are welcome! :p 
thanks

koalazak, 2018-12-04:
thanks @Zefau 
I have pending the cloud implementation for v2 firmwares
take a look if you want: #25

closed 2019-01-08 by koalazak