koalazak/dorita980 #78: Issue getting password

crashf, 2018-12-17:
Im having an issue getting the password. It gets all data except the password:
> node ./bin/getpassword.js "192.168.88.51"

Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
(node:18800) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
Robot Data:
{ ver: '3',
  hostname: 'Roomba-3162C22091522610',
  robotname: 'Dustin Beiber',
  ip: '192.168.88.51',
  mac: '80:C5:F2:BC:73:88',
  sw: '3.3.19',
  sku: 'R675020',
  nc: 0,
  proto: 'mqtt',
  cap: { ota: 1, eco: 1, svcConf: 1 },
  blid: '3162C22091522610' }

It grabs all that data then hangs forever. If I try it without "holding the dock button until it flashes" I get this:
> node ./bin/getpassword.js "192.168.88.51"

Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
(node:10988) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
Robot Data:
{ ver: '3',
  hostname: 'Roomba-3162C22091522610',
  robotname: 'Dustin Beiber',
  ip: '192.168.88.51',
  mac: '80:C5:F2:BC:73:88',
  sw: '3.3.19',
  sku: 'R675020',
  nc: 0,
  proto: 'mqtt',
  cap: { ota: 1, eco: 1, svcConf: 1 },
  blid: '3162C22091522610' }
Error getting password. Follow the instructions and try again.

Which I would expect

Roomba 675

koalazak, 2018-12-17:
seems like a problem of compatibility with your node.js version. Try using node 6 to grab the password

crashf, 2018-12-17:
Same issue with note 6.15.1

Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Robot Data:
{ ver: '3',
  hostname: 'Roomba-3162C22091522610',
  robotname: 'Dustin Beiber',
  ip: '192.168.88.51',
  mac: '80:C5:F2:BC:73:88',
  sw: '3.3.19',
  sku: 'R675020',
  nc: 0,
  proto: 'mqtt',
  cap: { ota: 1, eco: 1, svcConf: 1 },
  blid: '3162C22091522610' }
just hang here. Still hung. Is there any logs I can look at to see more info?

koalazak, 2018-12-17:
oh, I just realize that you own a 600 series robot. It is not supported by dorita980.
Follow this thread for updates: https://github.com/koalazak/dorita980/issues/71

closed 2018-12-17 by koalazak