koalazak/dorita980 #81: Problems with i7+

thorazine-12, 2019-04-14:
Hello,

I have i7+ with 1.6.6 firmware (EMEA). I'm able to do basic actions using Local API (start, pause, stop, dock); the only method that works is getWirelessStatus.
All others (getSys, getWeek, getPreferences etc.) return nothing.

Also Cloud access doesn't work - problem with resolving irobot.axeda.com :/
Result of myRobotViaCloud.getStatus():

```
pi@raspberrypi:~/Downloads/dorita980 $ node myapp.js
{ RequestError: Error: getaddrinfo ENOTFOUND irobot.axeda.com irobot.axeda.com:443
    at new RequestError (/home/pi/Downloads/dorita980/node_modules/request-promise-core/lib/errors.js:14:15)
    at Request.plumbing.callback (/home/pi/Downloads/dorita980/node_modules/request-promise-core/lib/plumbing.js:87:29)
    at Request.RP$callback [as _callback] (/home/pi/Downloads/dorita980/node_modules/request-promise-core/lib/plumbing.js:46:31)
    at self.callback (/home/pi/Downloads/dorita980/node_modules/request/request.js:185:22)
    at Request.emit (events.js:193:13)
    at Request.onRequestError (/home/pi/Downloads/dorita980/node_modules/request/request.js:881:8)
    at ClientRequest.emit (events.js:193:13)
    at TLSSocket.socketErrorListener (_http_client.js:397:9)
    at TLSSocket.emit (events.js:193:13)
    at emitErrorNT (internal/streams/destroy.js:91:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:59:3)
    at processTicksAndRejections (internal/process/task_queues.js:81:17)
  name: 'RequestError',
  message:
   'Error: getaddrinfo ENOTFOUND irobot.axeda.com irobot.axeda.com:443',
  cause:
   { Error: getaddrinfo ENOTFOUND irobot.axeda.com irobot.axeda.com:443
       at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:58:26)
     errno: 'ENOTFOUND',
     code: 'ENOTFOUND',
     syscall: 'getaddrinfo',
     hostname: 'irobot.axeda.com',
     host: 'irobot.axeda.com',
     port: 443 },
  error:
   { Error: getaddrinfo ENOTFOUND irobot.axeda.com irobot.axeda.com:443
       at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:58:26)
     errno: 'ENOTFOUND',
     code: 'ENOTFOUND',
     syscall: 'getaddrinfo',
     hostname: 'irobot.axeda.com',
     host: 'irobot.axeda.com',
     port: 443 },
  options:
   { method: 'GET',
     headers:
      { 'User-Agent': 'aspen%20production/2618 CFNetwork/758.3.15 Darwin/15.4.0',
        Accept: '*/*',
        'Accept-Language': 'en-us',
        'ASSET-ID': 'ElPaso@irobot!F1FFBE8B5FCE439CA080273553DFC57A' },
     uri:
      'https://irobot.axeda.com/services/v1/rest/Scripto/execute/AspenApiRequest?blid=F1FFBE8B5FCE439CA080273553DFC57A&robotpwd=:1:1552928199:fCYdULh6r9Zs14ES&method=getStatus',
     callback: [Function: RP$callback],
     transform: undefined,
     simple: true,
     resolveWithFullResponse: false,
     transform2xxOnly: false },
  response: undefined }
```

So what am I doing wrong?

koalazak, 2019-04-14:
seems liek the host irobot.axeda.com is down

thorazine-12, 2019-04-14:
but how can I resolve problem with getSys, getWeek etc. not working using local API?

koalazak, 2019-04-15:
Hi @thorazine-12 , can you run this code and share the output please?

```javascript
const dorita980 = require('dorita980');
  const myLocalRobot = new dorita980.Local('blid_here', 'password_here', 'ip_here');

  myLocalRobot.on('connect', () => { 
    console.log('connected'); 
    myLocalRobot.on('packetreceive', packet =>  {
      packet.payload = JSON.parse(packet.payload);
      console.log(JSON.stringify(packet, null, 2));
    });
  });
```

thanks

thorazine-12, 2019-04-15:
Sure I can, here is my config and results of script:

```
Robot Data:
{ ver: '3',
  hostname: 'iRobot-F1FFBE8B5FCE439CA080273553DFC57A',
  robotname: 'Roomba',
  robotid: 'F1FFBE8B5FCE439CA080273553DFC57A',
  ip: '192.168.1.12',
  mac: '50:14:79:01:5B:AA',
  sw: 'lewis+1.6.6+lewis-day-0-ota+29',
  sku: 'i755040',
  nc: 0,
  proto: 'mqtt',
  cap:
   { binFullDetect: 1,
     dockComm: 1,
     maps: 3,
     edge: 0,
     area: 1,
     eco: 1,
     multiPass: 2,
     pose: 1,
     team: 1,
     pp: 0,
     '5ghz': 1,
     prov: 3,
     sched: 1,
     svcConf: 1,
     ota: 2,
     log: 2 },
  blid: '.........................' }
Password=> .......................... <= Yes, all this string.
Use this credentials in dorita980 lib :)
```

```
pi@raspberrypi:~/Downloads/dorita980 $ node dorita-test.js
connected
{
  "cmd": "publish",
  "retain": false,
  "qos": 0,
  "dup": false,
  "length": 3172,
  "topic": "$aws/things/F1FFBE8B5FCE439CA080273553DFC57A/shadow/update",
  "payload": {
    "state": {
      "reported": {
        "batPct": 100,
        "batteryType": "F12432832R",
        "bbchg": {
          "nLithF": 0,
          "nChgOk": 21,
          "aborts": [
            0,
            0,
            0
          ],
          "smberr": 0
        },
        "bbchg3": {
          "estCap": 1826,
          "nAvail": 37,
          "hOnDock": 619,
          "avgMin": 71
        },
        "bbmssn": {
          "aCycleM": 48,
          "nMssnF": 1,
[... 562 lines cut ...]
        }
      }
    }
  }
}
{
  "cmd": "publish",
  "retain": false,
  "qos": 0,
  "dup": false,
  "length": 83,
  "topic": "wifistat",
  "payload": {
    "state": {
      "reported": {
        "signal": {
          "rssi": -52,
          "snr": 38,
          "noise": -90
        }
      }
    }
  }
}
```

koalazak, 2019-04-15:
ok the problem is those methods are waiting for properties in the state that are not there in i7+ robots (like `uiSwVer` and others). So never return the state. (they are present in 980 robots)
So, I recomend to you just use the `getRobotState(['fieldYouWant')` method to get the system properties you are looking for.
To be clear, those methods (getSys, getWeek, getPreferences) just wait for some properties in the state and then return that fields. For example `getWeek()` is waiting for `cleanSchedule` in the state and i7 robots has `cleanSchedule2` propertie but not `cleanSchedule`. To emulate getWeek() in i7 robots use `getRobotState(['cleanSchedule2')`.

I will try to fix this in next version.

thank you

closed 2019-04-15 by koalazak