koalazak/dorita980 #82: i7 / selective cleaning

thorazine-12, 2019-04-18:
Thanks to your support I managed to operate my i7 using dorita980 library.
But now I try to figure out how to start selective cleaning process (send i7 only to selected rooms), so I need to change _apiCall fuction somehow.
But the question is how to do it? I assume that there is no open-API for i7, so how can I figure out what API command will start this selective cleaning process?
I'd appreciate your comment about this.

koalazak, 2019-04-19:
Hi @thorazine-12,
exactly, there is no open-API for i7 so you need o sniff the trafic beatween the mobile app and the robot to see what is the command to do that.
If you found the command please reopen the issue and I will update the lib to add the command.
thank you!

closed 2019-04-19 by koalazak

briandlucas, 2019-06-25:
close but no luck...  the reported last command is:

`{"command":"start","initiator":"localApp","ordered":0,"pmap_id":"BYgLAfu2TTWnJHPksxd1ew","regions":[{"region_id":"15","region_name":"Laundry Room","region_type":"laundry_room"}],"time":1561487907,"user_pmapv_id":"190625T025451"}}}}`

later in the shadow update stream there are two pmaps (position maps?) each with a user pmap:
`"pmaps": [{"BYgLAfu2TTWnJHPksxd1ew": "190625T025451"}, {"iy0zJsSpTLK0MpsizdAc0g": "190625T172158"}],`

a command with two rooms selected for cleaning:
`"regions": [{"region_id": "14", "region_name": "Family Room", "region_type": "family_room"}, {"region_id": "15", "region_name": "Laundry Room", "region_type": "laundry_room"}]`

i also tried to prefix the command with state / desired / command as mentioned in another issue but also w/o success.

i would really like to get this to work... thoughts?

koalazak, 2019-06-26:
did you try with this?
```
const cmd = {command: 'start', time: Date.now() / 1000 | 0, initiator: 'localApp', "pmap_id":"BYgLAfu2TTWnJHPksxd1ew","regions":[{"region_id":"15","region_name":"Laundry Room","region_type":"laundry_room"}], "user_pmapv_id":"190625T025451"};
myRobotViaLocal.publish('cmd', JSON.stringify(cmd), function (e) {
  console.log(e);
});
```

this should start cleaning the Laundry Room

briandlucas, 2019-06-27:
so i actually wasn't running your code - was working on another project in another language... but i did fire it up this morning and now have a lead as to why my i7 just blinks at me with room-specific  commands but otherwise responds appropriately.  the [extra] length of the room-specific commands trips the MQTT remaining length into a second byte.  i never focused on the 7-bits for length and 1 bit as continuation flag - i was always under the 127 limit with simple start/stop/dock but the "regions" put me over the 1 byte limit.  thank you and your code for the clue!!