koalazak/dorita980 #83: Non-Human readable password

cyb3rg3cko, 2019-05-03:
I am attempting to use dorita980 on a Raspberry Pi3 with a group of i7s'. I don't have an internet connection or a way to use the roomba app to initially set up a network for it to attach to.  So I connected directly to the roomba while it is in access point mode and tried to use the getpassword command. The command returned all the robot information as expected however the password has many non-standard characters and causes an "Invalid or unexpected tolkens" error when attempting to use any script that requires the blid/password below is a copy of its response with the id's x-ed out but the password left as it showed. 

Robot Data:
{ ver: '3',
  hostname: 'iRobot-xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx',
  robotname: 'robot',
  robotid: 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx',
  ip: '0.0.0.0',
  mac: 'xx:xx:xx:xx:xx',
  sw: 'lewis+1.3.4+lewis-day-0-ota+18',
  sku: 'i755020',
  nc: 1,
  proto: 'mqtt',
  cap:
   { binFullDetect: 1,
     dockComm: 1,
     maps: 3,
     edge: 0,
     area: 1,
     eco: 1,
     multiPass: 2,
     pose: 1,
     pp: 0,
     '5ghz': 1,
     prov: 3,
     sched: 1,
     svcConf: 1,
     ota: 2,
     log: 2 },
  lps: 0,
  blid: 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx' }
Password=> t(�MP(�Mb�t(�MP(�M`��Ѧ <= Yes, all this string.
Use this credentials in dorita980 lib :)

koalazak, 2019-05-03:
Hi @deckar01,
seems like there is a new "v3" behavior in the password exchange process. (currently dorita980 are handling v1 and v2)
I dont own a i7 robot, so can you help me to debug the new process?
add this code to the first line into this callback https://github.com/koalazak/dorita980/blob/master/bin/getpassword.js#L95 to see the robot response:

```javascript
console.log('response length:', data.length);
console.log('response:');
console.log(data.toString());
console.log('--------------');
```
then run the normal process to get the password and share the output with me :)

thank you!

deckar01, 2019-05-03:
I think you @'ed the wrong user. I also don't own an i7. ;)

koalazak, 2019-05-03:
lol, sorry @deckar01. 
I mean @cyb3rg3cko

cyb3rg3cko, 2019-05-03:
I assume you meant to add it to the callback of the checkV2 function...
this was the response

Robot Data:
{ ver: '3',
  hostname: 'iRobot-xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx',
  robotname: 'robot',
  robotid: ' xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx  ',
  ip: '0.0.0.0',
  mac: '50:14:79:0A:F0:F4',
  sw: 'lewis+1.3.4+lewis-day-0-ota+18',
  sku: 'i755020',
  nc: 0,
  proto: 'mqtt',
  cap:
   { binFullDetect: 1,
     dockComm: 1,
     maps: 3,
     edge: 0,
     area: 1,
     eco: 1,
     multiPass: 2,
     pose: 1,
     pp: 0,
     '5ghz': 1,
     prov: 3,
     sched: 1,
     svcConf: 1,
     ota: 2,
     log: 2 },
  lps: 0,
  blid: ' xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx  ' }
response length: 47
response:
�.��;)t(�MP(�Mb� t(�MP(�M `� � Ѧ
--------------
Password=> t(�MP(�Mb� t(�MP(�M `� � Ѧ <= Yes, all this string.
Use this credentials in dorita980 lib :)

In addition, I added the following code to try and see what bytes were
being send

 console.log('Bytes:');
 var myBuffer = [];
 var buffer = new Buffer(data, 'utf16le');
 for (var i = 0; i < buffer.length; i++) {
 myBuffer.push(buffer[i]);
 }
 console.log(myBuffer);

Which responded with...

Bytes:
253, 255, 46, 0, 253, 255, 253, 255, 59, 0, 41, 0, 0, 0, 0, 0, 0, 0, 0, 0,
0, 0, 116, 0, 40, 0, 253, 255, 77, 0, 80, 0, 40, 0, 253, 255, 77, 0, 5, 0,
0, 0, 0, 0, 0, 0, 98, 0, 253, 255, 1, 0, 0, 0, 116, 0, 40, 0, 253, 255, 77,
0, 80, 0, 40, 0, 253, 255, 77, 0, 1, 0, 0, 0, 0, 0, 0, 0, 96, 0, 253, 255,
1, 0, 0, 0, 253, 255, 25, 0, 102, 4, 0, 0

Thanks for the quick response.

deckar01, 2019-05-03:
```
253, 255, 46, 0,
253, 255,
253, 255, 59, 0,  41, 0,   0, 0,  0, 0, 0, 0,  0, 0, 0, 0, 116, 0, 40, 0,
253, 255, 77, 0,  80, 0,  40, 0,
253, 255, 77, 0,   5, 0,   0, 0,  0, 0, 0, 0, 98, 0,
253, 255,  1, 0,   0, 0, 116, 0, 40, 0,
253, 255, 77, 0,  80, 0,  40, 0,
253, 255, 77, 0,   1, 0,   0, 0,  0, 0, 0, 0, 96, 0,
253, 255,  1, 0,   0, 0,
253, 255, 25, 0, 102, 4,   0, 0
```

cyb3rg3cko, 2019-05-06:
Just wanted to update this with what I found... After having my initial password response issue I decided to try and figure out a way to set the i7 units up using the iRobot mobile app. Like mentioned before I didn't have internet access on the LAN I wanted to hook them up to. I found that I could start the app on my phone while using my data plan and once the app was running connect to the robot and setup my LAN connection using the on-screen instructions. 

After the unit was setup and connected to the LAN I attempted to use the getpassword function which responded with an error indicating that the network was probably having issues with UDP packets. So I connected directly to the robot as I previously had done and tried the getpassword function again. The response this time included a valid password that I was able to use with testing the start, stop, and dock functions over the LAN (I'm assuming these don't use UDP).

My thoughts are maybe the getpassword function won't work properly unless initially configured with the mobile app.  I will probably reset one of the robots and look into this further but for now I will just set any unit up with the app first to get it going.