koalazak/dorita980 #88: Error while getting robot password

ilyabreev, 2019-07-16:
After running `npm run getrobotpwd <ip>` i got initial robot information including `blid` and then following error:

```
events.js:170
      throw er; // Unhandled 'error' event
      ^

Error: 1996351952:error:141A318A:SSL routines:tls_process_ske_dhe:dh key too small:../deps/openssl/openssl/ssl/statem/statem_clnt.c:2159:

Emitted 'error' event at:
    at emitErrorNT (internal/streams/destroy.js:91:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:59:3)
    at processTicksAndRejections (internal/process/task_queues.js:81:17)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! homebridge-roomba@1.0.1 getrobotpwd: `cd node_modules/dorita980 && npm install && node ./bin/getpassword.js "192.168.2.50"`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the homebridge-roomba@1.0.1 getrobotpwd script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-16T16_39_17_608Z-debug.log
```

koalazak, 2019-07-16:
homebridge-roomba? I think you are asking in the wrong project

closed 2019-07-16 by koalazak

ilyabreev, 2019-07-16:
It is used like standalone part of homebridge-roomba without any changes.
If I try run same command from this repo, the same problem appears.

koalazak, 2019-07-16:
`getrobotpwd` npm command does not exist on this library.

xXKeyleXx, 2019-07-30:
For me it worked to set the cipher of the TLS connection to `AES128-SHA256`

ilyabreev, 2019-07-30:
@xXKeyleXx Great! How can I set it? And what cipher suite do you mean? `TLS_RSA_WITH_AES_128_CBC_SHA`?

xXKeyleXx, 2019-07-30:
I modified the script like this:
```js
const client = tls.connect(8883, host, {rejectUnauthorized: false, ciphers: 'AES128-SHA256'}, () => {
    client.write(new Buffer(packet, 'hex'));
});
```

koalazak, 2019-07-30:
hey guys what node.js version do you have?

koalazak, 2019-07-30:
and what robot? firmware version?

xXKeyleXx, 2019-07-30:
I don't think the node version is the problem. It works fine on my desktop PC but I get this error on my Raspberry PI too.

Node: v12.7.0
Robot: e5
Firmware: 3.4.46-76

ilyabreev, 2019-07-30:
I think that problem is in the openssl version. Last openssl doesn't respect such ciphers as `AES128-SHA256`.

koalazak, 2019-08-02:
same problem reported in #91
I will work on a fix maybe hardcoding the cypher to AES128-SHA256
thanks @xXKeyleXx

koalazak, 2019-08-02:
hey guys, can you test dorita980@3.1.3 latest version? it should work now

koalazak, 2019-08-17:
@xXKeyleXx how are you?
hey, Can you run this `node -p process.versions | grep openssl` to let me know wich version of ssl is generating problems? I know you had this ssl problems on your box and I want to know which is the bad ssl version.
@ilyabreev can you share yours please?
thank you

ilyabreev, 2019-08-17:
```bash
pi@raspberrypi:~ $ node -p process.versions | grep openssl
  openssl: '1.0.2n'
```

xXKeyleXx, 2019-08-18:
`openssl: '1.1.1c'`