koalazak/dorita980 #90: New firmware 3.3.43-73

MarkyMarc35, 2019-07-19:
Hi,
I just got my roomba (671 wifi) and it has firmeware 3.3.43-73.
When i run the app to get the password, i don't get any data back and only a connection error is displayed. The roomba is turn on and it did make a sound when i pressed the home button for a few second and the wifi indicator does flash.

koalazak, 2019-07-19:
Hi @MarkyMarc35,

can you run this code and share the output please?

```
var dorita980 = require('dorita980');

dorita980.discovery((ierr, data) => {
  console.log(data);
});
```

thank you

koalazak, 2019-07-24:
seems like you have the mobile app open, Please close it and try again.
Or maybe you have a firewall blocking port 8883 on your roomba.

MarkyMarc35, 2019-07-25:
Turned off my phone to make sure the mobile app wasn't a factor, turned off my firewall as well and i get the same result.

koalazak, 2019-07-26:
can you `nmap` your robot ip?

closed 2019-08-02 by koalazak