koalazak/dorita980 #91: Issue getting password key too small

andijegg, 2019-08-02:
Hi,

I've problems to get the password. It seems it get the most infos about the Robot, but getting an error at the password

`key too small:../deps/openssl/openssl/ssl/statem/statem_clnt.c:2156:`

Robot 960
sw 2.4.6-3
node -v 10.16.1
npm -v 6.10.2

`Robot Data:
{ ver: '3',
  hostname: 'Roomba-3144411C00520730',
  robotname: 'Lisi',
  ip: '192.168.1.xxx',
  mac: 'F0:03:8C:C3:xx:xx',
  sw: 'v2.4.6-3',
  sku: 'R960040',
  nc: 0,
  proto: 'mqtt',
  cap:
   { pose: 1,
     ota: 2,
     multiPass: 2,
     pp: 1,
     binFullDetect: 1,
     langOta: 1,
     maps: 1,
     edge: 1,
     eco: 1,
     svcConf: 1 },
  blid: '3144411C00520730' }
events.js:174
      throw er; // Unhandled 'error' event
      ^

Error: 1995819472:error:141A318A:SSL routines:tls_process_ske_dhe:dh key too small:../deps/openssl/openssl/ssl/statem/statem_clnt.c:2156:

Emitted 'error' event at:
    at emitErrorNT (internal/streams/destroy.js:91:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:59:3)
    at process._tickCallback (internal/process/next_tick.js:63:19)
`

koalazak, 2019-08-02:
check this issue 👆 #88 for a temporary fix. 
Im working on a new version handling new ssl scenario

andijegg, 2019-08-02:
Thank you. I give it a try.

andijegg, 2019-08-02:
Hi,

doesn't worked for me.

I changed the script get password.js 

`
}

function checkV2 () {
  var sliceFrom = 13;
  discovery.getRobotPublicInfo(host, function (e, robotData) {
    console.log('Robot Data:');
    console.log(robotData);
  });
  const packet = 'f005efcc3b2900';
  const client = tls.connect(8883, host, {rejectUnauthorized: false, ciphers: 'AES128-SHA256'}, () => {
    client.write(new Buffer(packet, 'hex'));
  });

  client.on('data', function (data) {
    if (data.length === 2) {
      sliceFrom = 9;
      return;
    }
    if (data.length <= 7) {
      console.log('Error getting password. Follow the instructions and try again.');
    } else {
      console.log('Password=> ' + new Buffer(data).slice(sliceFrom).toString() + ' <= Yes, all this string.');
      console.log('Use this credentials in dorita980 lib :)');
    }
    client.end();
    process.exit(0);
  });

  client.setEncoding('utf-8');
}

console.log('Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button $

if (fversion === '1') {
  console.log('Then wait and look here...');
  checkV1(1);
} else {
  console.log('Then press any key here...');
  process.stdin.setRawMode(true);
  process.stdin.resume();
  process.stdin.on('data', checkV2);
}
`

Now I get the Following Error

`Error getting password. Follow the instructions and try again.`

koalazak, 2019-08-02:
how you install and how you run the command?
I just publish dorita980@3.1.3 with the fix. can you try new version?

andijegg, 2019-08-02:
First
```
npm install dorita980 --save
get-roomba-password <robotIP>
```

Now
```
git clone https://github.com/koalazak/dorita980.git
cd dorita980
npm install
npm run getpassword <robotIP>
```
Sure I can give it a try. How should I install it? via Update?

koalazak, 2019-08-02:
second option is fine. 
(I think first option will call your global instalation script and not the local directory version.)
Because if you dont have a global installation you should get a command not found error:

```
➜  npm install dorita980
+ dorita980@3.1.3
updated 1 package and audited 1705 packages in 1.875s
found 0 vulnerabilities

➜  get-roomba-password
sh: command not found: get-roomba-password
```
(I dont have a global installation ☝️ )

To use the local directory installation you also can do:

```
npm install dorita980
npm run getpassword
```

thank you!

koalazak, 2019-08-02:
(please verify you have the mobile app closed and press the home button not the Clean button)

andijegg, 2019-08-03:
In the Local Version I get a connection refused 

```
pi@mqtt:~ $ get-roomba-password 192.168.1.xx
Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
events.js:174
      throw er; // Unhandled 'error' event
      ^

Error: connect ECONNREFUSED 192.168.1.75:8883
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1106:14)
Emitted 'error' event at:
    at emitErrorNT (internal/streams/destroy.js:91:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:59:3)
    at process._tickCallback (internal/process/next_tick.js:63:19)
```

and otherwise I get the old error. I renamed the old dorita980 Folder an did a new git clone

```
pi@mqtt:~/dorita980 $ npm run getpassword 192.168.1.xx

> dorita980@3.1.3 getpassword /home/pi/dorita980
> node ./bin/getpassword.js "192.168.1.xx"

Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
(node:1929) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
Robot Data:
{ ver: '3',
  hostname: 'Roomba-3144411C005207xx',
  robotname: 'Lisi',
  ip: '192.168.1.xx',
  mac: 'F0:03:8C:C3:C9:xx',
  sw: 'v2.4.6-3',
  sku: 'R960040',
  nc: 0,
  proto: 'mqtt',
  cap:
   { pose: 1,
     ota: 2,
     multiPass: 2,
     pp: 1,
     binFullDetect: 1,
     langOta: 1,
     maps: 1,
     edge: 1,
     eco: 1,
     svcConf: 1 },
  blid: '3144411C005207xx' }
Error getting password. Follow the instructions and try again.
```

andijegg, 2019-08-03:
Mobile-App is closed via Task-manager

andijegg, 2019-08-03:
So now I've deinstalled dorita980 and reinstalled it global 

still the same Error

Error getting password. Follow the instructions and try again.

aimproxy, 2019-08-08:
I have the same error here:

```
Robot Data:
{ ver: '3',
  hostname: 'Roomba-3178480822035600',
  robotname: 'e5',
  ip: '192.168.1.94',
  mac: 'D0:C5:D3:BD:21:31',
  sw: '3.4.46',
  sku: 'e515840',
  nc: 0,
  proto: 'mqtt',
  cap: { ota: 1, eco: 1, svcConf: 1 },
  blid: '3178480822035600' }
Error getting password. Follow the instructions and try again.

```

koalazak, 2019-08-16:
I can't reproduce the error using same node, npm and firmware with a 980 robot:
```
sw 2.4.6-3
node -v 10.16.1
npm -v 6.10.2
```
And I don't own a e5 robot to debug in `3.4.46` firmware :/
try using diferent node.js versions like 6 or 8 please
and you can collaborate here to allow my access to a new firmware robot:
https://opencollective.com/dorita980

regards,

aimproxy, 2019-08-16:
@koalazak It's worked after closing all connection with the mobile app, my mom have the app in background! kkk You can close the issue

aimproxy, 2019-08-16:
@koalazak can you answer me about the other issue I create?

koalazak, 2019-08-16:
@aimproxy anwered.
One question for you: did you press only Home button to get the password? or in e5 robots you must press Home AND Spot buttons at the same time to make it works?

closed 2019-08-16 by koalazak

aimproxy, 2019-08-16:
@koalazak On the e5 robot, model nº e5158, sw nº 3.4.46, you must press the Home btn for seconds until the wifi icon turn green, this procedure can be done outside the base charging station! Let me know if you need more info for this particular robot.