koalazak/dorita980 #93: Intermittent hangs

TroyJ, 2019-08-09:
I'm getting intermittent hangs (hangs indefinitely).

For example, running this:

`var dorita980 = require('dorita980');
var username = process.argv[2];
var password = process.argv[3];
var ip = process.argv[4];
var myRobotViaLocal = new dorita980.Local(username, password, ip);

console.log('1');
myRobotViaLocal.on('connect', execute);
console.log('2');

function execute() {
  console.log('3');
  myRobotViaLocal.getRobotState()
      .then((actualState) => {
        console.log('4');
        console.log(actualState)
      })
      .catch(console.log);
}`

Gives the output:
`1
2`

Sometimes it works fine for days without doing this, and then it starts hanging. Rebooting the host machine doesn't help.

Any ideas on how to proceed?

koalazak, 2019-08-16:
can you share your node.js version and dorita980 version?
thanks

TroyJ, 2019-08-17:
npm dorita980 --version
6.9.0

node --version
v10.16.2

Zefau, 2019-08-17:
`npm dorita980 --version` is wrong and gives the npm version (which is 6.9.0 in your case).
Use `npm view dorita980 version`

TroyJ, 2019-08-18:
Oh, sorry... I'm new to npm.

npm view dorita980 version
3.1.3

koalazak, 2019-08-18:
looks a wifi issue with 960 robots (several reports with same error only with 960 robots) or maybe the same ssl problem fixed in `dorita980@3.1.4`
please upgrade to dorita980@3.1.4 and try again.

closed 2019-08-18 by koalazak

FrankBlackMG, 2019-09-10:
I have a 960 and have this same problem. I'm running dorita980 3.1.4. I can get my blid and password but no other functions work (never have), e.g. getting the weekly schedule or any status.

npm view dorita980 version
3.1.4

node --version
v10.16.3