koalazak/dorita980 #94: getpassword.js explanation about the tls connection

aimproxy, 2019-08-11:
Hi, I'm translating your code to Vala to use in (Home App)[https://github.com/manexim/home] for elementary OS, so we creating a library to translate ur code to Vala (libirobot)[https://github.com/manexim/libirobot].

I have some question about ur code, take a look below:

```
  const packet = 'f005efcc3b2900';
  var client = tls.connect(8883, host, {rejectUnauthorized: false, ciphers: 'AES128-SHA256'}, function () {
    client.write(new Buffer(packet, 'hex'));
  });
```

Here, nodejs etablish a TLS connection with cipher, whats is the value returned from `tls.connect` ?, ciphers should be `AES128` or `SHA256` ?, about `client.write` whats this function do? and `Buffer(packet, 'hex')` is to convert hex to bin?

koalazak, 2019-08-16:
Hi, 
nice!

`tls.connect ` returns a socket in where you can send bytes with `write` method.
Cipher is 'AES128-SHA256' (RSA key exchange and AES encryption) it is suite. You can't choose one Or other. They works in combination.
`Buffer(packet, 'hex')` creates a buffer using a hex string source.

the buffer contains bytes for a special packet in mqtt protocol
http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/mqtt-v3.1.1.html
http://www.steves-internet-guide.com/mqtt-protocol-messages-overview/

let me know if you need something else

closed 2019-08-16 by koalazak

aimproxy, 2019-08-16:
@koalazak Hi again, just one more thing, a question for a newbie programmer. There is a buffer in C like on NodeJS? and the variable packet is a hex string right? and that hex string will be encoded to a byte array right?