koalazak/dorita980 #95: EPROTO error while connecting

Zefau, 2019-08-17:
Hi there!

I have implemented an adapter using this library for an IoT platform (https://github.com/Zefau/ioBroker.roomba).

My users recently report that they are not able to connect to the Roomba anymore having the following error message:
```
{"errno":"EPROTO","code":"EPROTO","syscall":"write"}
```

They're config: Roomba 960 with FW 2.4.6-3 on Node.js v10.16.3
My config works though: Roomba 980 with FW v2.4.6-3 on Node.js v10.16.1

We have tried different stuff like https://forum.iobroker.net/topic/7657/irobot-roomba-adapter/232 and https://forum.iobroker.net/topic/7657/irobot-roomba-adapter/232 but it does not seem to work.

Do you have any ideas?
Thanks in advanced

koalazak, 2019-08-17:
can you run `node -p process.versions | grep openssl` in a box with the error? also in a box that it is working fine?

thanks

Zefau, 2019-08-17:
Both the one working and the one that fails with the error have the output
```
openssl: '1.1.1c',
```

koalazak, 2019-08-18:
mmmHmmM
can they try with this connection options?
```
var options = {
    port: 8883,
    clientId: user,
    rejectUnauthorized: false,
    protocolId: 'MQTT',
    protocolVersion: 4,
    ciphers: 'AES128-SHA256',
    clean: false,
    username: user,
    password: password
  };
```
(note the ciphers value)

koalazak, 2019-08-18:
all reports are for  `960` robot? or do you have other models with the problem?

Zefau, 2019-08-18:
Works fine for my 980 as well. Users reporting the issue having a 960. Have had no other users with other Roomba's reporting so far.

Zefau, 2019-08-18:
The suggested settings works! Great! Thanks. Can you implement that?

koalazak, 2019-08-18:
nice! implemented in dorita980@3.1.4
thanks for reporting

closed 2019-08-18 by koalazak