koalazak/dorita980 #98: Mission Coordinates to Latitude/Longitude

MediumFidelity, 2019-11-14:
Is there enough information from the Roomba APIs to determine the real world distance covered by moving some number of points?  They return `sqft `, so they have some concept of real work distances.

Assuming you know the starting position, you'd be able to plot a map on Google Maps or export KML.

koalazak, 2019-12-20:
@MediumFidelity Did you find something?
I don't know how much each `pos` number is worth (maybe 1 inch?)

koalazak, 2019-12-20:
the center wheel of my 980 has a 80mm aprox of circumference. Half black and half white. So the minimun resolution for the encoder should be 80/2 mm.
Probably they have 3 inchs of circumference (my measured in mm was not pretty good)
So probably 1 point of each `pos` increment values mean 1 or 1.5 inchs 🤷‍♂

koalazak, 2020-01-11:
@MediumFidelity any update on this?

closed 2021-09-13 by koalazak