koalazak/dorita980 #101: Unable to connect on Roomba 981

ycrumeyrolle, 2019-12-11:
Hi, I cannot connect to my Roomba 981 to get the password.
I have the following error message
```
Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
events.js:174
      throw er; // Unhandled 'error' event
      ^

Error: connect EHOSTUNREACH <my_roomba_IP>:8883
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1104:14)
Emitted 'error' event at:
    at emitErrorNT (internal/streams/destroy.js:82:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:50:3)
    at process._tickCallback (internal/process/next_tick.js:63:19)
```

The robot is on its dock, WIFI light flash, no app connected.

Is the 981 supported ?

koalazak, 2019-12-20:
not tested in 981. 
troubleshooting check list:
- Are you sure you are pressing HOME button and not Clean?
- All your mobile apps are closed?
- Did you configure the roomba for first time with the mobile app? (if not, you need to do that)
- Reset the robot pressing Clean for 10 seconds aprox. Wait for restart and try again
- Try other node.js version like 6 or 8

I will let the issue open waiting for someone that can confirm if it works on other 981 robots.

cdeharo, 2019-12-30:
Today I've tested a 981 with `docker run -it node sh -c "npm install -g dorita980 && get-roomba-password <robotIP>"` and it works perfectly.

closed 2020-01-11 by koalazak