koalazak/dorita980 #102: AES128-SHA256 cipher not supported by Electron

David-Hari, 2019-12-19:
See https://github.com/electron/electron/issues/20759.
Basically, when this library is using in an Electron context, it rejects the MQTT connection due to the cipher.
I found that changing the cipher allows it to connect, but I have to modify the library to do so.
I'm not sure of the consequences of using a different type of cipher, but I'd like it to either use a compatible one or allow me the option of passing my own.

David-Hari, 2019-12-21:
> So AES128-SHA256 works with all robots but other cyphers don't.
> Anyway I can add an environment variable to allow you to overwrite this value in dorita980 under your responsability :p.

A parameter passed to the constructor would be easier than an environment variable.

koalazak, 2019-12-21:
I know, but I don't want to polute the constructor just for a strage/edge case. Also the cipher is also used in get-password which is not under the constructor.
You can use .env files to make it more easy
Added `ROBOT_CIPHERS` env in `dorita980@3.1.7`. Also updated the documentation with notes about this.

closed 2019-12-21 by koalazak

David-Hari, 2019-12-21:
Cool, thanks

zachfeldman, 2019-12-22:
Very cool thanks @koalazak !